qmk/qmk_firmware #12900: [Bug] RGB_MATRIX_TYPING_HEATMAP is a security flaw

mikhirev, 2021-05-14:
The typing heatmap effect reveals passwords being typed on a keyboard to anyone who can see it and to surveillance cameras. Even if it is difficult to define the exact order of typed characters, the knowledge of pressed keys makes it very easy to brute force the password. I propose to disable this effect by default for all keyboards.

drashna, 2021-05-14:
Well, if they can see your keyboard well enough to see the rgb effects, then can probably tell what you're typing without the need of the rgb.  

Also, could be argued that any of the reactive modes have this issue too.

And by default, these modes need a define to be added to enable them. 

Also, you could turn off the effects completely with `RGB_TOG`. 

Which is to say, your concerns are valid, but I'm not sure how much of an issue this actually is.

mikhirev, 2021-05-14:
> Well, if they can see your keyboard well enough to see the rgb effects, then can probably tell what you're typing without the need of the rgb.

Usually I can be sure that someone staying near me and looking aside while I'm typing will not see my password. With this effect enabled, he still can look at the keyboard right after I finished typing and know which keys I pressed. Speaking about surveillance cameras, most of them have too low FPS to record typing process, but with this effect capturing a single frame can be enough to recover a password.

> Also, could be argued that any of the reactive modes have this issue too.

Well, may be. I'm new to QMK and don't know what you are speaking about.

> by default, these modes need a define to be added to enable them.

You need to define `DISABLE_RGB_MATRIX_TYPING_HEATMAP` to disable it, it is enabled by default.  Enabling it only when explicitly configured by user is what I propose. And it would be good to mention in the documentation that enabling this effect is insecure.

fauxpark, 2021-05-14:
What's wrong with just...not using it?

mikhirev, 2021-05-14:
@fauxpark I've answered your question just above.

fauxpark, 2021-05-14:
Not really... the typing heatmap effect is not the default animation, so to me disabling it outright wouldn't do a whole lot. Additionally, there is another requirement to have the heatmap effect compiled in: `RGB_MATRIX_FRAMEBUFFER_EFFECTS` must be defined, along with the absence of `DISABLE_RGB_MATRIX_TYPING_HEATMAP`.

These are the boards I can find with it defined (excluding user code as that's their own choice):
```
1upkeyboards/super16
aleblazer/zodiark
basekeys/trifecta
dztech/dz60rgb/v2
dztech/dz60rgb_ansi/v2
dztech/dz60rgb_wkl/v2
geekboards/macropad
handwired/colorlice
handwired/hnah40rgb
handwired/p65rgb
helix/rev3_4rows
helix/rev3_5rows
kbdfans/bella/rgb
kbdfans/bella/rgb_iso
kbdfans/kbdmini
latin17rgb
latin60rgb
le_chiffre
marksard/rhymestone/rev1
mechlovin/adelais/rgb_led/rev1
mechlovin/adelais/rgb_led/rev2
mechlovin/hannah60rgb/rev1
mechlovin/hannah60rgb/rev2
miller/gm862
monstargear/xo87/rgb
moonlander
mt64rgb
mt84
planck/ez
silverbullet44
sowbug/68keys
sowbug/ansi_tkl
yncognito/batpad
```
Not a ton compared to the amount of boards in QMK, but there are some popular ones like DZ60RGB and Planck EZ. Three of these are also macropads; good luck entering passwords with them, but on the other hand, the heatmap animation doesn't seem very useful on a macropad anyway.

drashna, 2021-05-15:
A better solution may be to add some randomless to the color increase for the heatmap.

mikhirev, 2021-05-15:
After looking at the list of effects more precisely I found more effects that are affected (that's what @drashna wrote about). So I edited the title and description.

kb-elmo, 2021-07-28:
just don't use it?

like it was already said those animations are neither activated by default nor are they the default animation that you see when first powering on the board after flashing it.

the user has to explicitely choose that animation.

blueyed, 2021-10-29:
Just for reference: they were changed to not being enabled by default in https://github.com/qmk/qmk_firmware/pull/13758 (after the issue had been created).

kb-elmo, 2021-10-29:
> Just for reference: they were changed to not being enabled by default in #13758 (after the issue had been created).

No.
This change was just for the default RGB animations.

The framebuffer animations were never enabled by default because they take up a lot of flash.
They were always explicit enable. Even before that PR.

closed 2022-08-29 by tzarc