qmk/qmk_firmware #12920: [Feature Request] Sleep mode for the lighting?

techmouse, 2021-05-16:
## Description
I don't know if I'm asking this in the right place or not, so forgive me if I'm not. But is there currently, or can there be in the future, a feature that will disable the lighting on a keyboard after a period of inactivity?

It won't turn the keyboard off, but just turn off the lighting after some amount of time has passed since the last key press event. Then once a key is pressed, the lighting will turn back on.

Obviously this would really help cut down on how much electricity is being used by the keyboard when the user is AFK. I know the lighting can be disabled manually, but I'm asking for an automatic feature.

fl3tching101, 2021-05-23:
You could do this pretty easily with `process_record_user` and a timer with `housekeeping_task_user` for a personal keyboard. I think it would be a good improvement for QMK in general as well though. **Note that the below is untested code, but you get the idea**

```
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    some_variable = timer_read32() + LIGHTING_TIMEOUT;
    rgblight_enable_noeeprom();
    return true;
}
```

```
void housekeeping_task_user(void) {
    if(timer_expired32(timer_read32(), some_variable)) {
        rgblight_disable_noeeprom();
    }
}
```

drashna, 2021-05-23:
RGB Matrix and LED Matrix have this already, but RGB Light does not.   But as mentioned by @fl3tching101, you can run your own code to do this, in the meanwhile.

techmouse, 2021-08-29:
Does this require any involvement from me?

closed 2022-04-17 by stale[bot]