qmk/qmk_firmware #12929: [Bug]

ryanm101, 2021-05-17:
Hi.

When I connect USB-C -> USB-C it doesn't seem to get power / connection.. (Macbook Pro / Dell XPS with linux) I've tried a load of cables all known good.

If I connect using USB-C -> USB-A and an adaptor it works fine.

If I connect to my windows machine USB-C -> USB-A (no adaptor needed) it works fine.

The cables work with the Discipline 65% which uses the ATMEGA32A.

Anyone else encountered this / have a solution?

tzarc, 2021-05-17:
Can you post a high-resolution photo of the board? I'm guessing it doesn't have the correct pull-down resistors on the board.

ryanm101, 2021-05-18:
I'm away for a few days, will disassemble and get photo on Sunday when I'm back

Xelus22, 2021-05-19:
https://kprepublic.com/products/bm16a-16-keys-custom-mechanical-keyboard-pcb-plate-programmed-numpad-layouts-qmk-firmware-with-rgb-bottom-underglow-alps-mx

From the website it is a hardware fault. Not a QMK error. 
The 5.1k pull-down resistors on the CC lines are not present. :(

Electro707, 2021-05-25:
[image]

There does seem to be 2 resistors next to the connector close to each other, but without a further closeup I can't tell if there are the 5.1k pulldown or something else.

tzarc, 2021-05-25:
Those would be USB D+/D- termination resistors, rather than CC pull-downs. At this point, all I can suggest is not to bother using this with a C-to-C cable.

ryanm101, 2022-04-16:
I've tried to add the pull downs but don't think I got the pins right

qnm, 2022-07-09:
Did you resolve this? I think I have a similar issue.

I'm using the Hasu HHKB controller (atmega32u4) with QMK, which works perfectly with a USB-C to USB-A cable, however `dmesg` is empty when I connect using USB-C to USB-C (AMD b550 board).

I'm confirmed I can use the same cable and board with my USB-C SSD.

tzarc, 2022-08-29:
Neither problem lies with QMK I'm afraid.

closed 2022-08-29 by tzarc