qmk/qmk_firmware #12969: Add Option to Randomize Dwell Time and Gap Time to Defeat Biometric Surveillance

0verk1ll, 2021-05-20:
JavaScript on websites can profile a user by measuring the dwell time and gap time of keystrokes. Randomizing this can prevent this form of surveillance.

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

## Description

<https://paul.reviews/behavioral-profiling-the-password-you-cant-change/> describes how JavaScript can be used to determine who is typing on a keyboard based on measuring their dwell and gap time of keystrokes. There is a Chrome extension, which randomizes keystroke timing to prevent this surveillance (https://chrome.google.com/webstore/detail/keyboard-privacy/aoeboeflhhnobfjkafamelopfeojdohk).

Implementing this protection against biometric surveillance in hardware would be more effective because it would work across all apps and websites visited by the user.

drashna, 2021-05-21:
Well, this would be pretty easy to implement for the most part.  And you could actually use the `process_record_user` function to do this. 

```c
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case KC_A ... KC_Z: 
            if (!record->event.pressed) {
                uint8_t r = rand() % 50;
                wait_ms(r);
            }
            break;
    }
    return true;
}
```

(untested, unverified code).

pierre-vo, 2021-06-03:
If only few people have it randomized, then that would make them identifiable as well :)
Maybe adding a fixed delay that is randomized at a certain interval (init at power up) would be enough? That way you are identifiable for this "session" but if you come back after a power cycle, they cannot link you to your previous visit?

wxyangf, 2021-06-10:
> ```c
> bool process_record_user(uint16_t keycode, keyrecord_t *record) {
>     switch (keycode) {
>         case KC_A ... KC_Z: 
>             if (!record->event.pressed) {
>                 uint8_t r = rand() % 50;
>                 wait_ms(r);
>             }
>             break;
>     }
>     return true;
> }
> ```

Problem with this code is that the max of rand is fixed. There's some black magic math that can let you account for this statistically.

A better solution combining with @pierre-vo's comment would be to:

a. delay random keystrokes
b. choose a random max-delay on boot of keyboard

closed 2021-10-11 by stale[bot]