qmk/qmk_firmware #13006: GMMK Pro RGB Matrix support

Gigahawk, 2021-05-25:
Copied from locked thread https://github.com/qmk/qmk_firmware/pull/12030

Tried posting this to /r/mechanicalkeyboards, mods don't seem to want to approve it, so I'm posting here instead.

Glorious claims to be "working" on adding RGB support to QMK, yet /u/GloriousThrall hasn't had any [real Github activity for over two months now](https://github.com/GloriousThrall).

Now, I would be happy to give them the benefit of the doubt and say that implementing RGB is complicated...
that is, if I didn't manage to do it myself in like 2 days.

Proof: https://youtu.be/KGloqwgCuhg

If you want, you can try it out yourself here (at your own risk, I take no responsibility for bricked boards): https://drive.google.com/file/d/1NpuYt0k1-uu1wfUAfCMfcrMQBPrqtqGZ/view?usp=sharing

Note: In the video some keys don't light up, this may be due to:

- I mapped some LEDs incorrectly in the firmware
- I damaged the second LED driver when desoldering it to probe some pads.

Edit: seems to be a bit of both, the mapping for printscreen (next to function row) is probably mapped incorrectly, and I think I have bad solder connections for the enter and down arrow keys. 

If you see some keys not light up, please let me know, as that means I have to spend some time figuring out what address those LEDs are mapped to.

The keymap used for this build is here: https://pastebin.com/QjcKTtK5

You will see that I have disabled a lot of keys for this demo, and I have decided to withhold the source code for this until:

- I get a personal apology for being ignored and banned from the Glorious discord
- The community gets a proper apology/explanation for how Glorious has handled this situation, pinned to /r/Glorious until all remaining issues have been resolved
    - What has actually been going on with RGB support? This took me like 2 days to write, it should not have taken more than a week or two even if you were busy doing other things
    - Why was a community member required to make the pull request for VIA support after the board was released? This is something that could have easily been done alongside the initial pull request for QMK. 
    - What chip is going to be used in the Q3 batch? You say you've identified a chip, yet chose not to name it even though the community probably would have helped to implement it into QMK. Have you already secured enough chips for production? If you haven't, are they even still available?
- Consent from Awinic or G-SPY to release the source code
    - The LED driver chip being used is a proprietary chip made by Awinic, presumably for your contractor G-SPY, the datasheet for this is not public, and I'm not releasing source code until they are aware that publicly available code will contain info from this datasheet.

_Originally posted by @Gigahawk in https://github.com/qmk/qmk_firmware/issues/12030#issuecomment-848182821_

Gigahawk, 2021-05-25:
Paging @GloriousThrall, @pinpox, @van1337, @pascalpfeil,

tzarc, 2021-05-25:
As I said, we share your frustrations -- but you were asked to keep things on-topic. As I wasn't explicit, for on-topic I meant adding support for RGB to QMK for the GMMK Pro.

I'm going to give you a bit of time to edit the post to keep things relevant -- the vast majority of this issue is irrelevant to that cause.

closed 2021-05-25 by Gigahawk

pinpox, 2021-05-25:
@Gigahawk any way to reach you outside of GitHub ?

Gigahawk, 2021-05-25:
Reddit at /u/Gigahawk

Twitter at @Gigahawk515