qmk/qmk_firmware #13012: [Feature Request]  Add Augio keyboard to the configurator

jeschroeder, 2021-05-26:
## Feature Request Type

- [ ] Core functionality
- [x ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

There are layouts in the main GitHub (https://github.com/danforthewin/Augio) but it would be great if it were in the configurator. I really love this version of a 40% +numpad and with all files needed to make one listed it would be an even better beginner board if it were hosted in the Configurator. I particularly would love it as I am trying to work on my second as a project with my landlords 12 year old and they really love the interactive layout.

fauxpark, 2021-05-26:
#9440

drashna, 2021-05-30:
To be added to the configurator, it has to be added to the repo, first.

balaklavier, 2021-06-08:
Here's a first attempt at adding the Augio, @jeschroeder if you're comfortable building the firmware mind trying  my branch and seeing if it works on your keyboard before I open a PR? https://github.com/balaklavier/qmk_firmware/tree/augio

jeschroeder, 2021-06-09:
That worked on regular and debug mode. @balaklavier thanks for working on this.

balaklavier, 2021-06-09:
#13148

balaklavier, 2021-06-11:
When someone gets a chance I'd appreciate a review of #13148 as a first-time contributor I think there's also an additional approval needed to get the workflow automation to run.