qmk/qmk_firmware #13050: [Feature Request] Autoshift status callback

TweetyDaBird, 2021-05-30:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

I'm basically looking for a function similar to the is_combo_enabled() callback/function, using it to display the autoshift status on the OLED display.

rgoulter, 2021-06-02:
The naming isn't consistent, and the functions aren't listed in the docs, but there is `bool     get_autoshift_state(void)` which does what you want. Per the header file: https://github.com/qmk/qmk_firmware/blob/93496c8364369f72e2db90c3c59dfa3ecc8657b4/quantum/process_keycode/process_auto_shift.h

TweetyDaBird, 2021-06-08:
Well then, I guess we should just put it in the docs instead...!

> The naming isn't consistent with the name you suggest, and the functions aren't listed in the docs, but there is `bool get_autoshift_state(void)` which does what you want. Per the header file: https://github.com/qmk/qmk_firmware/blob/93496c8364369f72e2db90c3c59dfa3ecc8657b4/quantum/process_keycode/process_auto_shift.h

Well, then I guess it's just a matter of updating the docs instead. Many thanks!