qmk/qmk_firmware #13059: [Tracker] for Next ChibiOS-Contrib update

elfmimi, 2021-05-31:
While it is not clearly stated in the [timetable](https://beta.docs.qmk.fm/developing-qmk/breaking-changes/breaking_changes), at the time of writing, last update  https://github.com/qmk/qmk_firmware/pull/13056  was done on 2021 May 31, or 2021 Jun 1 depending of TZ.

See https://github.com/qmk/qmk_firmware/issues/12843 for some more context.

Here I request the same for the next coming cycle. and hopefully another chance of update in between for develop branch.

We can use this issue to track what changes to expect.

tzarc, 2021-05-31:
Already done. `develop` is updated.

elfmimi, 2021-05-31:
@tzarc Thank you for your hard work as always. I'm sure it will enable supports for many interesting MCUs.
I have plans for future. and also I have some pull-requests laying around still un-merged. out of laziness, my bad.

elfmimi, 2021-08-29:
at **2021Q3** breaking change https://github.com/qmk/qmk_firmware/pull/14196

[qmk/ChibiOS-Contrib @ ae35428](https://github.com/qmk/ChibiOS-Contrib/tree/ae3542876c928b352dcdaab537dc44ac73d1f45c)  → [qmk/ChibiOS-Contrib @ 4568901](https://github.com/qmk/ChibiOS-Contrib/tree/4568901a91e9bef78ea96a7a83e8150fe1f7353a)

No change for ChibiOS [qmk/ChibiOS @ 413e39c](https://github.com/qmk/ChibiOS/tree/413e39c5681d181720440f2a8b7391f581788d7b)