qmk/qmk_firmware #13088: [Bug] RGB_DISABLE_WHEN_USB_SUSPENDED not working for dz60rgb v2 using macOS

ichbindu, 2021-06-03:
## Describe the Bug

Keyboard LEDs don't turn off when the computer goes to sleep despite`RGB_DISABLE_WHEN_USB_SUSPENDED` set to `true` using the setup described below. Built using the vanilla firmware without any changes (`qmk compile -kb dztech/dz60rgb/v2 -km default`).

## System Information

 - Keyboard: dz60rgb v2 hotswap
 - Operating system: macOS 11.2.1 (Macbook 2019, hibernate mode 3 (default))
 - AVR GCC version: 9.3.0
 - ARM GCC version: 10.2.1 20201103
 - QMK Firmware version: 0.12.52
 - Any keyboard related software installed? N/A

vinorodrigues, 2021-08-15:
I can confirm this is the case as-well with my KDB67 keyboard:

## SysInfo
* Keyboard:  KBDFans KDB67 Mk.II V3 *{AVR (ATmega32U4)}*
* Operating System:  macOS 10.15.7
* AVR GCC version:  Homebrew AVR GCC 9.3.0_3
* QMK Firmware version:  0.13.34
* Any keyboard related software installed?  Karabiner-Elements 13.5.0 *{but not used in sleep modes}*

vinorodrigues, 2021-09-03:
This seems to be fixed now with pull request #14084

closed 2024-10-13 by drashna