qmk/qmk_firmware #13099: [Bug]  Arch/Manjaro: 'MILC' object has no attribute '_subcommand'

nstetter, 2021-06-03:
## Describe the Bug

When trying to compile any firmware QMK fails with the error: 
```
~/qmk_firmware❯ qmk compile -kb keebio/iris/rev3 -km default

<class 'AttributeError'>
☒ 'MILC' object has no attribute '_subcommand'
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/milc/milc.py", line 467, in __call__
    return self.__call__()
  File "/usr/lib/python3.9/site-packages/milc/milc.py", line 472, in __call__
    return self._entrypoint(self)
  File "/home/norman/qmk_firmware/lib/python/qmk/decorators.py", line 20, in wrapper
    if cli.config_source[cli._subcommand.__name__]['keyboard'] != 'argument':
AttributeError: 'MILC' object has no attribute '_subcommand'
```

The same when directly invoking make:
```
~/qmk_firmware ❯ make keebio/iris/rev3:default  
QMK Firmware 0.13.2
Making keebio/iris/rev3 with keymap default

☒ 'MILC' object has no attribute '_subcommand'
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/milc/milc.py", line 467, in __call__
    return self.__call__()
  File "/usr/lib/python3.9/site-packages/milc/milc.py", line 472, in __call__
    return self._entrypoint(self)
  File "/home/norman/qmk_firmware/lib/python/qmk/decorators.py", line 20, in wrapper
    if cli.config_source[cli._subcommand.__name__]['keyboard'] != 'argument':
AttributeError: 'MILC' object has no attribute '_subcommand'
build_keyboard.mk:104: 'AttributeError'>: No such file or directory
make[1]: *** No rule to make target ''AttributeError'>'.  Stop.
Make finished with errors
make: *** [Makefile:530: keebio/iris/rev3:default] Error 1
```

## System Information

 - Keyboard:  Iris (but error is independet of build-target)
   - Revision (if applicable):  rev3
 - Operating system: Manjaro Linux
 - QMK Firmware version: 0.13.2
 - Python version: 3.9.5

- Output of qmk doctor: 
  ```
  ~/qmk_firmware ❯ qmk doctor
  Ψ QMK Doctor is checking your environment.
  Ψ QMK home: /home/norman/qmk_firmware
  Ψ Detected Linux.
  Ψ All dependencies are installed.
  Ψ Found arm-none-eabi-gcc version 10.3.0
  Ψ Found avr-gcc version 8.3.0
  Ψ Found avrdude version 6.3
  Ψ Found dfu-util version 0.10
  Ψ Found dfu-programmer version 0.7.2
  Ψ Submodules are up to date.
  Ψ QMK is ready to go
  ```

## Additional Context

pierre-vo, 2021-06-03:
Same here, `pip install milc --upgrade` should get you milc >= 1.4 which has cli._subcommand.

Requirements.txt might need to be updated to reflect that.

nstetter, 2021-06-07:
@pierre-vo, thanks for the hint. Upgrading milc resolves the issue.

closed 2021-06-07 by nstetter