qmk/qmk_firmware #13131: Preonic Rev3 with a center green led after plug in

Acelya-9028, 2021-06-07:
Hello, I use a preonic /rev3 and qmk configurator to download the firmware and flash it.

<details>
  <summary>keymap</summary>
  
```
{
  "version": 1,
  "notes": "",
  "documentation": "\"This file is a QMK Configurator export. You can import this at <https://config.qmk.fm>. It can also be used directly with QMK's source code.\n\nTo setup your QMK environment check out the tutorial: <https://docs.qmk.fm/#/newbs>\n\nYou can convert this file to a keymap.c using this command: `qmk json2c {keymap}`\n\nYou can compile this keymap using this command: `qmk compile {keymap}`\"\n",
  "keyboard": "preonic/rev3",
  "keymap": "preonic_rev3_layout_ortho_5x12_mine",
  "layout": "LAYOUT_ortho_5x12",
  "layers": [
    [
      "KC_GRV",
      "KC_1",
      "KC_2",
      "KC_3",
      "KC_4",
      "KC_5",
      "KC_6",
      "KC_7",
      "KC_8",
      "KC_9",
      "KC_0",
      "KC_DEL",
      "KC_TAB",
      "KC_Q",
      "KC_W",
      "KC_E",
      "KC_R",
      "KC_T",
      "KC_Y",
      "KC_U",
      "KC_I",
      "KC_O",
      "KC_P",
      "KC_BSPC",
      "KC_ESC",
      "KC_A",
      "KC_S",
      "KC_D",
      "KC_F",
      "KC_G",
      "KC_H",
      "KC_J",
      "KC_K",
      "KC_L",
      "KC_SCLN",
      "KC_ENT",
      "KC_LSFT",
      "KC_Z",
      "KC_X",
      "KC_C",
      "KC_V",
      "KC_B",
      "KC_N",
      "KC_M",
      "KC_COMM",
      "KC_DOT",
      "KC_SLSH",
      "KC_RSFT",
      "MO(5)",
      "KC_LCTL",
      "KC_LALT",
      "KC_LGUI",
      "MO(3)",
      "KC_SPC",
      "KC_SPC",
      "MO(4)",
      "KC_LEFT",
      "KC_DOWN",
      "KC_UP",
      "KC_RGHT"
    ],
    [
      "KC_GRV",
      "KC_1",
      "KC_2",
      "KC_3",
      "KC_4",
      "KC_5",
      "KC_6",
      "KC_7",
      "KC_8",
      "KC_9",
      "KC_0",
      "KC_DEL",
      "KC_TAB",
      "KC_Q",
      "KC_W",
      "KC_F",
      "KC_P",
      "KC_G",
      "KC_J",
      "KC_L",
      "KC_U",
      "KC_Y",
      "KC_SCLN",
      "KC_BSPC",
      "KC_ESC",
      "KC_A",
      "KC_R",
      "KC_S",
      "KC_T",
      "KC_D",
      "KC_H",
      "KC_N",
      "KC_E",
      "KC_I",
      "KC_O",
      "KC_ENT",
      "KC_LSFT",
      "KC_Z",
      "KC_X",
      "KC_C",
      "KC_V",
      "KC_B",
      "KC_K",
      "KC_M",
      "KC_COMM",
      "KC_DOT",
      "KC_SLSH",
      "KC_RSFT",
      "MO(5)",
      "KC_LCTL",
      "KC_LALT",
      "KC_LGUI",
      "MO(3)",
      "KC_SPC",
      "KC_SPC",
      "MO(4)",
      "KC_LEFT",
      "KC_DOWN",
      "KC_UP",
      "KC_RGHT"
    ],
    [
      "KC_GRV",
      "KC_1",
      "KC_2",
      "KC_3",
      "KC_4",
      "KC_5",
      "KC_6",
      "KC_7",
      "KC_8",
      "KC_9",
      "KC_0",
      "KC_DEL",
      "KC_TAB",
      "KC_QUOT",
      "KC_COMM",
      "KC_DOT",
      "KC_P",
      "KC_Y",
      "KC_F",
      "KC_G",
      "KC_C",
      "KC_R",
      "KC_L",
      "KC_BSPC",
      "KC_ESC",
      "KC_A",
      "KC_O",
      "KC_E",
      "KC_U",
      "KC_I",
      "KC_D",
      "KC_H",
      "KC_T",
      "KC_N",
      "KC_S",
      "KC_ENT",
      "KC_LSFT",
      "KC_SCLN",
      "KC_Q",
      "KC_J",
      "KC_K",
      "KC_X",
      "KC_B",
      "KC_M",
      "KC_W",
      "KC_V",
      "KC_Z",
      "KC_RSFT",
      "MO(5)",
      "KC_LCTL",
      "KC_LALT",
      "KC_LGUI",
      "MO(3)",
      "KC_SPC",
      "KC_SPC",
      "MO(4)",
      "KC_LEFT",
      "KC_DOWN",
      "KC_UP",
      "KC_RGHT"
    ],
    [
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TILD",
      "KC_EXLM",
      "KC_AT",
      "KC_HASH",
      "KC_DLR",
      "KC_PERC",
      "KC_CIRC",
      "KC_AMPR",
      "KC_ASTR",
      "KC_LPRN",
      "KC_RPRN",
      "KC_TRNS",
      "KC_DEL",
      "KC_F1",
      "KC_F2",
      "KC_F3",
      "KC_F4",
      "KC_F5",
      "KC_F6",
      "KC_UNDS",
      "KC_PLUS",
      "KC_LCBR",
      "KC_RCBR",
      "KC_DQUO",
      "KC_TRNS",
      "KC_F7",
      "KC_F8",
      "KC_F9",
      "KC_F10",
      "KC_F11",
      "KC_F12",
      "KC_TRNS",
      "KC_TRNS",
      "KC_HOME",
      "KC_END",
      "KC_PIPE",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "MO(6)",
      "KC_MNXT",
      "KC_VOLD",
      "KC_VOLU",
      "KC_MPLY"
    ],
    [
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_GRV",
      "KC_1",
      "KC_2",
      "KC_3",
      "KC_4",
      "KC_5",
      "KC_6",
      "KC_7",
      "KC_8",
      "KC_9",
      "KC_0",
      "KC_TRNS",
      "KC_DEL",
      "KC_F1",
      "KC_F2",
      "KC_F3",
      "KC_F4",
      "KC_F5",
      "KC_F6",
      "KC_MINS",
      "KC_EQL",
      "KC_LBRC",
      "KC_RBRC",
      "KC_QUOT",
      "KC_TRNS",
      "KC_F7",
      "KC_F8",
      "KC_F9",
      "KC_F10",
      "KC_F11",
      "KC_F12",
      "KC_TRNS",
      "KC_TRNS",
      "KC_PGUP",
      "KC_PGDN",
      "KC_BSLS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "MO(6)",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_MNXT",
      "KC_VOLD",
      "KC_VOLU",
      "KC_MPLY"
    ],
    [
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_F13",
      "KC_F14",
      "KC_F15",
      "KC_F16",
      "KC_TRNS",
      "KC_LEFT",
      "KC_DOWN",
      "KC_UP",
      "KC_RGHT",
      "KC_TRNS",
      "KC_QUOT",
      "KC_TRNS",
      "KC_F17",
      "KC_F18",
      "KC_F19",
      "KC_F20",
      "KC_TRNS",
      "KC_F21",
      "KC_F22",
      "KC_F23",
      "KC_F24",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_PSCR",
      "KC_MPRV",
      "KC_MNXT",
      "KC_MPLY"
    ],
    [
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "RESET",
      "DEBUG",
      "RGB_TOG",
      "RGB_M_P",
      "RGB_HUI",
      "RGB_HUD",
      "RGB_SAI",
      "RGB_SAD",
      "RGB_VAI",
      "RGB_VAD",
      "KC_TRNS",
      "KC_TRNS",
      "ANY(AU_ON)",
      "ANY(AU_OFF)",
      "ANY(MI_ON)",
      "ANY(MI_OFF)",
      "KC_TRNS",
      "KC_TRNS",
      "DF(0)",
      "DF(1)",
      "DF(2)",
      "TO(5)",
      "KC_TRNS",
      "KC_TRNS",
      "ANY(MU_ON)",
      "ANY(MU_OFF)",
      "ANY(MU_MOD)",
      "ANY(TERM_ON)",
      "ANY(TERM_OFF)",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS",
      "KC_TRNS"
    ]
  ],
  "author": ""
}
```
</details>

I like use the plain rgb mode to have a a slight diffuse color through my acrylic case, but when I plug in my preonic, led 0, the one in the center, is plain green like (0, 255, 0), even if the color of other leds is plain red and I have to force plain rgb mode with my `RGB Mode P` key or use `RGB Toggle` key two times to have the 9 of the right color.

I don't know if it's a bug but it doesn't seem normal and I don't know how to fix it. Have you an idea? Thanks.

zvecr, 2021-06-09:
Please could you try to reproduce now that #13142 has been merged.

Acelya-9028, 2021-06-09:
Thank you very much.

Hum, yes and no. Yes it's fixed because now all 9 leds are the same color but there is a new behavior that I had not seen before. Before: When I plug in, leds take ~2s before lighting up. Now: When I plug in, leds instantly flash very briefly green if sat is at minimal then take ~2s before the final lighting up, perfect with all choosen colors. Green flash is very fast, with a low intensity, and occurs when saturation is minimal.

leds are set to hue red, sat max so color is red -> no green flash
leds are set to hue red, sat mini so color is white -> green flash

I didn't notice any change in effect if I change leds brightness.

So this is the result of my test, thank you very much for the merge.

closed 2021-10-11 by stale[bot]