qmk/qmk_firmware #13174: Discussion: Should QMK seek a valid USB vendor and product ID through pid.codes?

BalzGuenat, 2021-06-10:
Most boards and the new-board-template use `0xFEED` as the USB vendor ID. While this works fine, it's technically "strictly prohibited" by the [_USB Implementers Forum_](https://www.usb.org/developers).

Unfortunately, registering as a vendor costs $5000 annually. Fortunately, there is [pid.codes](https://pid.codes/) which owns a valid VID and will assign a valid PID to any open-source project that needs one. My understanding is that this project would qualify.

I suggest QMK seeks a PID from pid.codes to use in its new-board-template, as well as for any existing boards to use freely if they so desire.

kb-elmo, 2021-07-26:
This wouldn't really be much useful for QMK.
The 0xFEED is just the default.

Every keyboard that implements support for VIA or Vial needs to have a different VID and PID.
And this is pretty much every new keyboard nowadays. 
Meaning the 0xFEED isn't used for new keyboards anyways most of the time.

xyzz, 2021-07-26:
FWIW, that's not true for vial boards, they can share the same vid/pid. It's actually how I've got all mine setup since I bought a pair from a shady place. USB-IF really doesn't want you to sublicense a VID space you purchased - whether free or not. It's unclear how legal this is, and whether just squatting an unused VID (the way most via boards do it) is any worse.

tzarc, 2021-07-26:
XAP won't care, either. VIA is the only real problem, in that respect.
I'll add it to my todo list as far as criteria for XAP "completion" is concerned.

BalzGuenat, 2021-07-27:
@xyzz If I understand the explanation of pid.codes correctly, the contract of their VID is old and from a time when sublicensing wasn't contractually prohibited yet. That's how they get away with it.

closed 2022-01-09 by stale[bot]