qmk/qmk_firmware #13176: [Bug] Split keyboard backlight breathing only working on one side

gabrieldonadel, 2021-06-11:
## Describe the Bug

Recently I've built a hand-wired Iris keyboard and I was trying to get the backlight breathing to work, the problem is that when I enable it, only one side takes on the effect (the master side).

PS: I'm currently using a serial connection over a TRRS
 

## System Information

 - Keyboard:  Handwired Iris
 - Operating system: Windows 10
 - AVR GCC version: 8.4.0
 
 - QMK Firmware version: 0.13.2

 - Any keyboard related software installed? No 

## Additional Context

https://user-images.githubusercontent.com/11707729/121613217-636a0480-ca32-11eb-9503-337fcb604ee1.mp4

thattolleyguy, 2021-06-16:
I'm experiencing the same issue. I'm not a C dev but as far as I can tell, it looks like the BL duty is not sent across from master to slave. I'm looking into how to pass that across effectively but may take a bit to figure out.

tzarc, 2021-06-16:
Sounds about right, actually. Might see if I can add it to #11930.

thattolleyguy, 2021-06-16:
@tzarc That would be awesome. My C is rusty or I'd give it a try. Happy to help test if needed.

gabrieldonadel, 2021-06-16:
@tzarc I can definitely help testing it as well if needed

gabrieldonadel, 2021-07-19:
@tzarc was this fixed on your PR?

gabrieldonadel, 2021-11-15:
@tzarc any updates on this?

tzarc, 2021-11-15:
Zero updates. Low priority compared to other things I'm afraid.

osleg, 2021-12-02:
Breath works on both halfs of my dactyl manuform using QMK 0.15

gabrieldonadel, 2021-12-02:
> Breath works on both halfs of my dactyl manuform using QMK 0.15

Thanks for the update, gonna try updating it today

smeury, 2023-11-28:
Is this still open? I have the same issue.

drashna, 2024-10-13:
Closing due to age, and being resolved.

smeury, if you're still having this issue, please open a new issue.

closed 2024-10-13 by drashna