qmk/qmk_firmware #13178: [Bug] Preonic rev3 lights and chime not working with newly compiled firmware

kpk47, 2021-06-11:
## Describe the Bug

My Drop Preonic rev3 lights and sound behave erratically after flashing with the latest firmware. The keyboard does not chime when powering on but does when switching layers or when I press the reset button. Music mode works as normal. The keyboard's lights do not turn on. The keyboard also doesn't persist which layer is in use when unplugged and plugged back in. The keyboard still works when plugged into my MacBook Air and iPad Air, but it no longer works with my PS4. The PS4 is able to recognize that a keyboard is plugged in, but it cannot receive input.

None of these bugs appear when flashed with the precompiled firmware from qmk.tv/keyboards, but they do appear when flashed with firmware compiled from Github. I have tried building firmware with QMK Configurator with the default keymap and on OSX with a clean repo clone, and the bugs appear in both situations. My build environment is OSX 11.4 with qmk's latest release from Homebrew.

## System Information

 - Keyboard:  Preonic
   - Revision (if applicable):  rev3
 - Operating system: OSX 11.4
 - AVR GCC version: 

avr-gcc (Homebrew AVR GCC 8.4.0_2) 8.4.0
 - ARM GCC version: 

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
 - QMK Firmware version:

0.13.5
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

I tried syncing back a couple of weeks to start build bisection and was not able to find a build that did not exhibit the bugs described. That makes me thing the root cause is in my build environment rather than qmk itself. 

I have previously flashed this keyboard with custom firmware without issue, but I built that firmware on Windows instead of OS X. That was about 2 months ago.

kpk47, 2021-06-12:
I bisected commits and learned that this bug was introduced in the May 2021 Breaking Change.

3araht, 2021-06-13:
> I bisected commits and found that this bug was introduced in the May 2021 Breaking Change.

I have similar symptoms showing that it's not updating the LED status when I use RGB MATRIX before and after somewhere there.
At least the merge with 495d47b7d2, committed on May 15, 2021, had no problem. 
However, I'm facing the problem at least after the merge with 9dcd03b6ca, committed on June 8, 2021.

3araht, 2021-06-13:
I narrowed it down that the symptom occurs from QMK Firmware 0.12.52.

`git checkout 0.12.51; make clean; make chromatonemini:led:flash`
This is OK. LEDs are refreshed as expected.

`git checkout 0.12.52; make clean; make chromatonemini:led:flash`
This fails. It won't refresh the status of each LED.

I used a keyboard which is under a pull request.
https://github.com/qmk/qmk_firmware/pull/12768

3araht, 2021-06-13:
Strange. It depends on the keyboard. I have a few keyboards that are having the problem, while some are not.
For instance, giabalanai (also under a pull request https://github.com/qmk/qmk_firmware/pull/10125) is working fine with the latest firmware environment (QMK Firmware 0.13.5.)
I can't find the differences yet.

drashna, 2021-06-16:
By chance, are you enabling LTO?

3araht, 2021-06-16:
> By chance, are you enabling LTO?

No, I don't. No match below ran on the top folder of all my copied keyboard codes:
`grep -r LTO .`

3araht, 2021-07-11:
Good news, I tested the software of chromatonemini with QMK Firmware 0.13.19 today and found that LEDs are refreshed as expected.

3araht, 2021-07-11:
Uhh. Bad news, when I tested with a different keyboard (bandominedoni), it showed the same symptom that the LEDs are not refreshed...

3araht, 2021-07-17:
> Good news, I tested the software of chromatonemini with QMK Firmware 0.13.19 today and found that LEDs are refreshed as expected.

Oh, not again... It happens with chromatonemini https://github.com/qmk/qmk_firmware/pull/13516 with QMK Firmware 0.13.22.
LEDs are NOT refreshed again. Once it turns on, it keeps the color. No animation is moving at all.
Same symptom with the above versions/keyboards when I jumped into this thread.

drashna, 2024-10-13:
Chime can be fixed with `#define AUDIO_INIT_DELAY`

As for leds, is this still an issue?