qmk/qmk_firmware #13190: please help

BobbyBoyGaming, 2021-06-12:
hello I need help with the xd002, I created a keymap using QMK but this ATTiny MCU 2-button keyboard can only be flashed via the "micronucleus CLI exe file" found in https://github.com/micronucleus/micronucleus

I created a keymap, of the appropriate size, according to QMKSYS, then I installed the zadig drivers according to the micronucleus GitHub instructions (windows 10). then I start windows CMD as admin and cd into the micronucleus exe and enter the command 'micronucleus --run <firmware_file_name>

But I keep getting "please plug in the device"
I tried unplugging and plugging (also with the reset feature while holding the button nearest to micro-USB socket while plugging in the PCB)

I tried doing the "micronucleus --erase-only" command first but I still get "please plug-in device..."

Does anybody know what is going on? I only have Windows 10 at the moment.
[image]

BobbyBoyGaming, 2021-06-12:
I am so confused.... So many questions. I have never used "make" before. Where do I type that into? "make xd002:default:flash" goes into Git Bash ? (it says "make command not recognized")

Also, how did you manage to put the xd002 into "DFU mode"? I tried the reset method (hold button nearest micro-USB socket while plugging-in PCB), however, this is not working for me. Do you know how to put the xd002 into DFU mode? Could you please explain this to me? Thank you!

[image]

zvecr, 2021-06-12:
In that screenshot you are not really following the instructions i posted. You dont need to `make install` as thats what the `cp` is doing to the prebuilt binaries.

The flash commands assume you have followed <https://docs.qmk.fm/#/newbs_getting_started>.

Reset method was as you describe.

closed 2021-10-11 by stale[bot]