qmk/qmk_firmware #13191: Error right after qmk setup

AstroDrabb, 2021-06-12:
I just order the Anavi Macropad8 and while I am waiting for it to arrived I wanted to setup my own keymap.  After pulling from git I run "qmk setup"

    $ qmk setup
    Ψ Found qmk_firmware at /home/jim/Downloads/qmk_firmware.
    Ψ QMK Doctor is checking your environment.
    Ψ QMK home: /home/jim/Downloads/qmk_firmware
    Ψ Detected Linux.
    Ψ All dependencies are installed.
    Ψ Found arm-none-eabi-gcc version 11.1.0
    Ψ Found avr-gcc version 11.1.0
    ⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
    Ψ Found avrdude version 6.3
    Ψ Found dfu-util version 0.10
    Ψ Found dfu-programmer version 0.7.2
    Ψ Submodules are up to date.
    Ψ QMK is ready to go, but minor problems were found

So only thing is a warning about avr-gcc that is available on Arch Linux is to new.

Next I try to build the default keymap:
```
$ qmk setup
Ψ Found qmk_firmware at /home/jim/Downloads/qmk_firmware.
Ψ QMK Doctor is checking your environment.
Ψ QMK home: /home/jim/Downloads/qmk_firmware
Ψ Detected Linux.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 11.1.0
Ψ Found avr-gcc version 11.1.0
⚠ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.10
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```
Does anyone know what is going on?  Thanks for the help.

rgoulter, 2021-06-13:
@AstroDrabb looks like you've copy-pasted the same code snipped twice. What was the actual error listing from `qmk compile -kb anavi/macropad8 -km default`?

AstroDrabb, 2021-06-13:
> @AstroDrabb looks like you've copy-pasted the same code snipped twice. What was the actual error listing from `qmk compile -kb anavi/macropad8 -km default`?

My bad.  Here it is:
````
$ qmk compile -kb anavi/macropad8 -km default
<class 'AttributeError'>
☒ 'MILC' object has no attribute '_subcommand'
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/milc/milc.py", line 467, in __call__
    return self.__call__()
  File "/usr/lib/python3.9/site-packages/milc/milc.py", line 472, in __call__
    return self._entrypoint(self)
  File "/home/jim/Downloads/qmk_firmware/lib/python/qmk/decorators.py", line 20, in wrapper
    if cli.config_source[cli._subcommand.__name__]['keyboard'] != 'argument':
AttributeError: 'MILC' object has no attribute '_subcommand'
````
pulled r5.a0848659 from:
https://github.com/qmk/qmk_firmware.git

python-nose2 / 0.10.0-1
python-pep8-naming / 0.11.1-1
yapf / 0.31.0-1
python-milc / 1.3.0-1
qmk / 0.0.48-1
python-appdirs / 1.4.4-3
python-argcomplete / 1.12.1-1
python-colorama / 0.4.4-3
python-dotty-dict / 1.3.0-3
python-hidapi / 0.10.1-1
python-hjson / 3.0.2-3
python-jsonschema / 3.2.0-4
python-milc / 1.3.0-1
python-pygments / 2.9.0-1
python-pyusb / 1.1.1-1

rgoulter, 2021-06-13:
Hmm. Looking at that `lib/python/qmk/decorators.py` file shows it was last changed in #13046, which mentions #13099 which is a duplicate of this.

#13099 https://github.com/qmk/qmk_firmware/issues/13099#issuecomment-854229526 suggests to run:
```
pip` install milc --upgrade
```

closed 2022-01-09 by stale[bot]