qmk/qmk_firmware #13194: [Bug] QMK MSYS can’t clone repository

Evotron, 2021-06-12:
I cannot clone QMK’s qmk_firmware with QMK MSYS by running `qmk setup` because it fails due to the following:
`☒ Could not run 'git clone --recurse-submodules --branch=master https://github.com/qmk/qmk_firmware C:/qmk_firmware': UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf3 in position 12: invalid continuation byte`

## Describe the Bug

QMK MSYS does not clone the QMK qmk_firmware github repository when using the `qmk setup` command.
## System Information

 - Keyboard:  Ergodox EZ
   - Revision (if applicable):  
 - Operating system: Windows 10
 - AVR GCC version:  avr-gcc.exe (GCC) 8.4.0

 - ARM GCC version: arm-none-eabi-gcc.exe (GCC) 10.1.0

 - QMK Firmware version: can’t install it

 - Any keyboard related software installed? 
   - [ Yes] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

QMK MSYS can’t clone the repository. I’ve tried cloning it to different drives and folders, but nothing works.

jbanaszczyk, 2021-06-14:
Try to use `qmk setup -y`. Do it twice

jbanaszczyk, 2021-06-14:
Duplicate #13112

oco9oco, 2021-09-08:
try `LANG = "en_US.UTF-8"` before `qmk setup`, then you can find the missing trident-like QMK icon appears.

closed 2022-08-29 by tzarc