qmk/qmk_firmware #13195: [Bug] Problems with sending KC_MUTE in complex layouts

gagarski, 2021-06-12:
`KC_MUTE` keycode is sent fine when defined directly in `keymaps` array however there are issues when sending it using `LT` macro, `MT` macro or tap-dance feature.

## Describe the Bug

Let's try this test layout for Geekboards Macropad V2:

```c
#include QMK_KEYBOARD_H
#include <string.h>

enum custom_keycodes {
  CUSTOM_MUTE = SAFE_RANGE,
  CUSTOM_A
};

enum tap_dance {
    TD_MUTE = 0,
    TD_A
};

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
    [0] = LAYOUT_ortho_2x4(
        LT(1, KC_MUTE), KC_MUTE, CUSTOM_MUTE, TD(TD_MUTE),
        LT(1, KC_A),    KC_A,    CUSTOM_A,    TD(TD_A)
    ),
    [1] = LAYOUT_ortho_2x4(
        KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS,
        KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS
    )
};

void tap_dance_mute_finished(qk_tap_dance_state_t *state, void *user_data) {
    if (state->count == 1) {
        if (state->interrupted || !state->pressed) {
            tap_code16(KC_MUTE);
        }
    }
}

void tap_dance_a_finished(qk_tap_dance_state_t *state, void *user_data) {
    if (state->count == 1) {
        if (state->interrupted || !state->pressed) {
            tap_code16(KC_A);
        }
    }
}

qk_tap_dance_action_t tap_dance_actions[] = {
    [TD_MUTE] = ACTION_TAP_DANCE_FN_ADVANCED(NULL, tap_dance_mute_finished, NULL),
    [TD_A] = ACTION_TAP_DANCE_FN_ADVANCED(NULL, tap_dance_a_finished, NULL)
};

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case CUSTOM_MUTE:
            if (record->event.pressed) {
                register_code16(KC_MUTE);
            } else {
                unregister_code16(KC_MUTE);
            }
            break;
        case CUSTOM_A:
            if (record->event.pressed) {
                register_code16(KC_A);
            } else {
                unregister_code16(KC_A);
            }
            break;
    }
    return true;
}
```

Basically we have four mute keys at top row and four A keys at the bottom.

Bottom row works as expected: all four keys type 'a' when pressed, first one and fourth one with little delay (also as expected).

Strange things happen with the top row: second and third keys work as expected (mute the volume), first one mutes the volume only if tapped twice which is quite strange. Fourth (mute with tap-dance) does not work at all.

A little delay seems to fix the issue with tap-dance:
```
void tap_dance_mute_finished(qk_tap_dance_state_t *state, void *user_data) {
    if (state->count == 1) {
        if (state->interrupted || !state->pressed) {
            register_code16(KC_MUTE);
            wait_ms(200);
            unregister_code16(KC_MUTE);
        }
    }
}
```

There is no explanation at all given in QMK docs for this behavior.

## System Information

 - Keyboard:  Geekboards Macropad V2
 - Operating system: Arch Linux, Windows 10 (as a keyboard users)
 - ARM GCC version: arm-none-eabi-gcc (Arch Repository) 11.1.0
 - QMK Firmware version: 0.13.5

## Additional Context

Interestingly double-tap behavior with LT is documented in printed keyboard documentation. @ruddy17, maybe you have some idea why this happens?

ruddy17, 2021-06-15:
LT() and double tap behavior is strange indeed. It should work from a single tap, but I decided to leave it as it is for now. I'll take a look into other cases described above.

gagarski, 2021-06-15:
BTW, case with `CUSTOM_MUTE` can also be broken:
```
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch (keycode) {
        case CUSTOM_MUTE:
            if (record->event.pressed) {
                tap_code16(KC_MUTE);
            }
            break;
        case CUSTOM_A:
            if (record->event.pressed) {
                tap_code16(KC_A);
            }
            break;
    }
    return true;
}
```

This way `KC_A` is sent and `KC_MUTE` is not (so this can be reproduced without enabling tap-dance)

gagarski, 2021-07-01:
Not sure if it's a good idea to have two versions of firmware behaving differently, but OK. If you do nothing, there.still.might.be a situation with two different behaviours (if this will be somehow fixed in QMK core). Maybe it's better to do it sooner and in expected manner though.

closed 2024-10-13 by drashna