qmk/qmk_firmware #13218: [Bug] Compile error with USE_I2C and RGB_MATRIX_SPLIT

y-k-m, 2021-06-16:
## Describe the Bug

1. config `RGB_MATRIX_ENABLE = yes` and `RGB_MATRIX_DRIVER = WS2812`
2. define `USE_I2C` and `RGB_MATRIX_SPLIT`
3. Compile error occurs.

## System Information

 - Keyboard:  Original
   - https://github.com/yoyo-real/qmk_firmware/tree/feature/YK-4
 - Operating system: 
   - Windows
 - AVR GCC version: 
   - avr-gcc.exe (GCC) 8.4.0
 - ARM GCC version: 
   - arm-none-eabi-gcc.exe (GCC) 10.1.0
 - QMK Firmware version:
   - 0.13.9

## Additional Context

```
Compiling: quantum/process_keycode/process_magic.c                                                  [OK]
Compiling: quantum/process_keycode/process_grave_esc.c                                              [OK]
Compiling: quantum/split_common/transport.c                                                        quantum/split_common/transport.c: In function 'transport_master':
quantum/split_common/transport.c:169:5: error: cannot convert to a pointer type
     i2c_writeReg(SLAVE_I2C_ADDRESS, I2C_RGB_MATRIX_START, (void *)rgb_matrix_config, sizeof(i2c_buffer->rgb_matrix), TIMEOUT);
     ^~~~~~~~~~~~
quantum/split_common/transport.c: In function 'transport_slave':
quantum/split_common/transport.c:225:5: error: cannot convert to a pointer type
     memcpy((void *)i2c_buffer->rgb_matrix, (void *)rgb_matrix_config, sizeof(i2c_buffer->rgb_matrix));
     ^~~~~~
quantum/split_common/transport.c:225:5: error: cannot convert to a pointer type
quantum/split_common/transport.c:225:5: error: null argument where non-null required (argument 1) [-Werror=nonnull]
quantum/split_common/transport.c:225:5: error: null argument where non-null required (argument 2) [-Werror=nonnull]
cc1.exe: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:436: .build/obj_yk_4_default/quantum/split_common/transport.o] エラー 1
Make finished with errors
make: *** [Makefile:530: yk_4:default] エラー 1
```

drashna, 2021-06-17:
I know what the issue is, and I'll have a fix for this shortly.  But basically. it needs to use `rgb_matrix_config.raw` instead of `rgb_matrix_config`

drashna, 2021-06-17:
Looks like a bunch of missing `&` mostly, and incorrectly copied memory.  🗡️

Nice catch!