qmk/qmk_firmware #13226: [Bug] nix-shell environment doesn't have the qmk command

IndianBoy42, 2021-06-17:
I think this is more about my confusion with how the Nix based setup works. i ran the following in the root of the repository:

```
nix-shell --pure --command "make"
```

I get the following errors

```
QMK Firmware 0.13.5
which: no qmk in (/nix/store/fgbzvd4c6nly9m4dpczrxybdpkm8mnk3-bash-interactive-4.4-p23/bin:/nix/store/8nwgidwa623jkz09wn0rjvcs07jjval3-patchelf-0.12/bin:/nix/store/pqiwg1jw0s7qzdi4m9xbps2jb98fsxbx-gcc-wrapper-10.2.0/bin:/nix/store/xrb5qsxpvhpm0irr9ykfqpvj8a6sk4xh-gcc-10.2.0/bin:/nix/store/b6p5cspsidkc2bpljhwz7d0nbmgrx5z0-glibc-2.32-37-bin/bin:/nix/store/9fkdn82nzxgkxyax3m9qdh2knj7apaa5-coreutils-8.32/bin:/nix/store/r7v3x5a68a4mxc2kh04cmyzx7xq00w2g-binutils-wrapper-2.35.1/bin:/nix/store/xdii8qvch5h8chyp0z2is2qzky565w68-binutils-2.35.1/bin:/nix/store/b6p5cspsidkc2bpljhwz7d0nbmgrx5z0-glibc-2.32-37-bin/bin:/nix/store/9fkdn82nzxgkxyax3m9qdh2knj7apaa5-coreutils-8.32/bin:/nix/store/sy9dql3x20ff9q56ciwssqkn9g1yvrsi-clang-tools-11.1.0/bin:/nix/store/jf2hbsl6fsd72dsgcx01qd2rgxbx2cmy-dfu-programmer-0.7.2/bin:/nix/store/ayjxybi9c8b8rh47412lzx6x2k0wa39x-dfu-util-0.10/bin:/nix/store/7v1xa3528d5frr55d3dhl991zx7by169-diffutils-3.7/bin:/nix/store/17awbq23ljcfl9m7rcfg6fffb1fcwg3x-git-2.30.1/bin:/nix/store/lz1f0gyfvw5q7lm1w4wzx8nj4kmhrilw-python3-3.8.8-env/bin:/nix/store/1rjmx2y4bzww1w3hjv4faxdfp0sxknm9-python3.8-poetry-1.1.5/bin:/nix/store/l9zijw30zdfmd2c98j30bllp8qmafyih-niv-0.2.19-bin/bin:/nix/store/65v9f16grqymgddjxdsji1m7rw83jlgx-avr-binutils-wrapper-2.35.1/bin:/nix/store/j4v4idxv0n1xi4vnlzpywqxpqnrl858f-avr-binutils-2.35.1/bin:/nix/store/9a3v0qsnaaj6j9k1dvvfdnivfg784jf2-avr-libc-2.0.0-avr/bin:/nix/store/9fkdn82nzxgkxyax3m9qdh2knj7apaa5-coreutils-8.32/bin:/nix/store/869g320dka91mv4z4348b33gxkgwzdbr-avr-stage-final-gcc-debug-wrapper-8.4.0/bin:/nix/store/mbpwbx86vsm7wxc9yc4v496qgf985id1-avr-stage-final-gcc-debug-8.4.0/bin:/nix/store/9a3v0qsnaaj6j9k1dvvfdnivfg784jf2-avr-libc-2.0.0-avr/bin:/nix/store/9fkdn82nzxgkxyax3m9qdh2knj7apaa5-coreutils-8.32/bin:/nix/store/9a3v0qsnaaj6j9k1dvvfdnivfg784jf2-avr-libc-2.0.0-avr/bin:/nix/store/lfsm7qyqlzkkgx69z5f52yncyc0nl9hh-avrdude-6.3/bin:/nix/store/b86fbamwswj596c1md5f0sh0a5wsbz8x-which-2.21/bin:/nix/store/hwss0kjy2iq5d39bhic4yk114m6vshsw-gcc-arm-embedded-10-2020-q4-major/bin:/nix/store/ligp0jnj36q4qxwi5mrjwqwipy1aigma-teensy-loader-cli-2.1.20191110/bin:/nix/store/9fkdn82nzxgkxyax3m9qdh2knj7apaa5-coreutils-8.32/bin:/nix/store/7nx8fqzpi5nsbi07md1qb1gbf9fscf9i-findutils-4.7.0/bin:/nix/store/7v1xa3528d5frr55d3dhl991zx7by169-diffutils-3.7/bin:/nix/store/ij7k48dhdrda13gcp3lbpi507b2awnmr-gnused-4.8/bin:/nix/store/ia1dn8zyi07phxkx02c0wwqc7v7c54nl-gnugrep-3.6/bin:/nix/store/9sscma6kgp35kb70bi8pms35ppk7j35r-gawk-5.1.0/bin:/nix/store/hxpycgfpdnmnc9iwxakxr037k0bvdq97-gnutar-1.32/bin:/nix/store/89kh3d91b9i9mqh39fj6xdff1sw81m9f-gzip-1.10/bin:/nix/store/vw1lcig1bkncmsmhif27h8hq1gaxldhg-bzip2-1.0.6.0.2-bin/bin:/nix/store/n7vham7g9lgwb7v83sk5nl7akk3zspvq-gnumake-4.3/bin:/nix/store/w65ydq10abi813fi7d5j7afdcrxj3aqq-bash-4.4-p23/bin:/nix/store/lsr2s27gbqxny7layxjab55dlqzrl1v2-patch-2.7.6/bin:/nix/store/sg3h64z6a7mxmap8zgrhhjscpri5b5vn-xz-5.2.5-bin/bin)
ERROR: Can not run "qmk" command!

 Please run util/qmk_install.sh to install all the dependencies QMK requires.

make: *** [Makefile:512: all:all] Error 1
error: Recipe `sh` failed on line 23 with exit code 2
```

Is this no longer a supported way of installing qmk? I can't seem to find Nix mentioned in the documentation.

IndianBoy42, 2021-06-17:
`nix-shell -p qmk` also doesn't seem to work for me, for the same reason, `Can not run "qmk" command`

rgoulter, 2021-06-17:
Try with this patch here: https://github.com/qmk/qmk_firmware/compare/develop...rgoulter:nix-shell-update-poetry-and-add-hidapi?expand=1

IndianBoy42, 2021-06-17:
This works, although I am worried about the deprecation warning of 'bin/qmk'

rgoulter, 2021-06-17:
Good, I opened #13227 then.

The more relevant issue is #12109, I suppose; where `nix-shell` *will* have to somehow have the `qmk` Python package installed.

sigprof, 2021-09-01:
This particular problem (missing `qmk` inside the nix-shell environment) was fixed in #12855 (although the part which also fixes some more obscure things like `qmk pytest` just recently arrived in #13316).

closed 2021-09-01 by sigprof