qmk/qmk_firmware #13272: [Feature Request] Provide a way to modify the `led_state` consumed by `led_update_kb` from `led_update_user`

t-8ch, 2021-06-20:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Currently when a user wants to override the handling of standard LED indicators from `led_update_user` they have to return `false` from this handler.
This means that the `_kb` handler is completely skipped and the user has to reimplement all of its functionality they want to actually still use.
While this is not a lot of code to carry around this could be optimized.

One easy way would be for `led_update_user` to take `led_state` as pointer which could be modified and then used in its modified form by `led_update_kb`

t-8ch, 2021-10-31:
/unstale

This has a PR open

t-8ch, 2022-07-18:
/unstale

tzarc, 2022-10-06:
Closed by #14452.

PR #14452 (2022-10-06): quantum: led: split out led_update_ports() for customization of led behaviour
Allow implementors of led_update_user() to modify the current led_state
but still use the default functionality of led_update_kb().
This can be used to reuse one of the standard LEDs for a different
usecase without having to reimplement led_update_kb().

Fixes #13272

This is still missing updates to the docs and keymaps.
Will do those after a first round of feedback.

- [x] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition o [... cut]
closed 2022-10-06 by tzarc