qmk/qmk_firmware #13283: DZ65RGB V3 not able to reset via escape key [Bug]

noelvissers, 2021-06-21:
When having the PCB in bootloader mode (will show as usb drive), flashing the compiled .bin file from qmk (and reconnecting the pcb) will disable the reset via escape option. 

## Describe the Bug

When having the PCB in bootloader mode (will show as usb drive), flashing the compiled .bin file from qmk (and reconnecting the pcb) will disable the reset via escape option.  Trying to reset the PCB via this method (holding esc while plugging in the usb c cable) will hold the pcb into a 'off' state. When the esc key is released, the keyboard will turn on as normal. 

After this, the keyboard is fully usable. I noticed this will however reset the RGB to default, which means some memory is erased, but it will not show as usb device anymore.

The problem here is that the only way to flash new firmware is to unscrew the pcb and use the reset switch on the back. This resets everything and allows you to access the board as USB drive again.

## System Information

 - Keyboard: DZ65RGB 
   - Revision (if applicable):  v3

## Additional Context

Installation process I used:
- Made layout using https://config.qmk.fm/ 
- Put uC in bootloader via holding esc and plugging in the usb cable
- replace FLASH.bin by the compiled .bin from https://config.qmk.fm/
- Eject usb via windows
- Re-plug usb cable
- Like said before, it now works as normal, with the keymapping made using https://config.qmk.fm/, but it is impossible to go into bootloader mode again unless the hardware button on the back of the PCB is pressed.

drashna, 2021-06-21:
> reset via escape option

You mean the bootmagic option? 

If so, does the RESET keycode work, either?

noelvissers, 2021-06-21:
> > reset via escape option
> You mean the bootmagic option?

With the default firmware from kbdfans you can hold the esc key (or top left key) while plugging in the usb cable to go into bootloader mode. This makes the keyboard show up as USB drive to simply copy and paste the .bin file to it. This is the new way of flashing apparently.

> If so, does the RESET keycode work, either?
Haven't tried, but also saw someone with the same issue in qmk discord, who said that did not work for him.

drashna, 2021-06-21:
Yeah, that's bootmagic. 

And if that and the reset keycode doesn't work, then the issue is that the default (lufa/atmel dfu) reset method doesn't work for the lufa-ms and a custom implementation is needed. 

I'm glad that kbdfans testing all this out before shipping...

KrDimitrov, 2021-06-21:
So i managed to make a hacky macro so i can go in to bootloader easier, since using the hardware reset button is not really that optimal.

Here is the code that successfully jumps to bootloader, at least on my board:
```cpp
#include <util/delay.h>
#include <avr/interrupt.h>

// ...

SEND_STRING("bootloader reset");

_delay_ms(2000);
// turn off interrupts
cli();
UDCON = 1;
USBCON = (1<<FRZCLK);  // disable USB

// turn off more stuff
// https://www.pjrc.com/teensy/jump_to_bootloader.html
UCSR1B = 0;
EIMSK = 0; PCICR = 0; SPCR = 0; ACSR = 0; EECR = 0; ADCSRA = 0;
TIMSK0 = 0; TIMSK1 = 0; TIMSK3 = 0; TIMSK4 = 0; UCSR1B = 0; TWCR = 0;
DDRB = 0; DDRC = 0; DDRD = 0; DDRE = 0; DDRF = 0; TWCR = 0;
PORTB = 0; PORTC = 0; PORTD = 0; PORTE = 0; PORTF = 0;
_delay_ms(2000);
// jump to bootloader address
asm volatile("jmp 0x7000");
```

You can make a custom macro with `process_record_user` and have a custom key execute it! 

Hope this turns out to be useful for someone else :)

fauxpark, 2021-06-22:
@KrDimitrov the `RESET` keycode should be doing, essentially, that.

I've flashed the Mass Storage bootloader onto a test board (really just a Pro Micro with a matrix attached): https://github.com/qmk/lufa/tree/ce10f7642b0459e409839b23cc91498945119b4d/Bootloaders/MassStorage

Playing around with it, Bootmagic Lite seems to work, as does the `RESET` keycode and ejecting the drive to re-enter QMK. So, I can't really reproduce this at all.

There is a trampoline at `0x6000` (the start of the aux bootloader section) to `0x7000` (the start of the primary bootloader section), so if you set `BOOTLOADER_SIZE` to either 8192 or 4096, entering the bootloader from within QMK should work. Until fairly recently though, this trampoline would have been at `0x6800` (for a total size of 6kB rather than 8kB):
[image]

I am wondering where this trampoline actually is on KBDfans' version of the bootloader...but in my testing it should not matter too much as long as the firmware is small enough to not overlap the `0x6000` - `0x6800` section of flash. The CPU should keep stepping through until it hits the `jmp`.

It may also be helpful to find out what fuses the MCUs are being flashed with. If those are fine too... I think this may have to be chalked up to KBDfans simply sourcing bad chips, which may also be the cause of the problems their boards were having with `atmel-dfu` that prompted them to switch.

MrRobot245, 2021-06-22:
The reset/ESC does seem to do some kind of reset, but it doesn't mount the filesystem. The board doesn't accept input until re-plugged in once the ESC / reset code is entered though.

Used the included VIAL firmware and their bootloader reset worked, as well as the initial firmware with holding ESC

fauxpark, 2021-06-22:
> The reset/ESC does seem to do some kind of reset, but it doesn't mount the filesystem. The board doesn't accept input until re-plugged in once the ESC / reset code is entered though.

Can you see it in Device Manager? When I hit `View->Devices by container` I get this:
[image]

MrRobot245, 2021-06-23:
Nothing comes up under mass storage when I put it into reset.
I can see device manager refresh, but there isn't a tree for mass storage

It looks like it times itself out too. Waiting about 30s the board is able to be typed on again.

datafx, 2021-07-26:
This board is basically the same as the new kbd67 mkiirgb v3 that I'm trying to port myself.  The issue is that they used the 6k boot loader.  Set "BOOTLOADER_SIZE = 6144" in rules.mk and that should work.

MrRobot245, 2021-07-26:
> Set "BOOTLOADER_SIZE = 6144" in rules.mk and that should work.

This is working for me, thank you!

H3lli0n, 2023-03-13:
Hi, I've encounter the same issue with latest VIA firmware of KBD67 MKII ISO RGB. I posted on VIA Git today but seems from this side, isnt't it ?

zvecr, 2023-11-14:
Fixed by #13729.

closed 2023-11-14 by zvecr