qmk/qmk_firmware #13344: [Feature Request] Documentation for using simple switches with JOYSTICK_ENABLE

skullydazed, 2021-06-25:
## Feature Request Type

- [x] Documentation

## Description

WIth https://github.com/qmk/qmk_firmware/pull/4226 by @a-chol we added gamepad support and support for analog joysticks. However, not all game controllers use analog joysticks. The documentation says that you can use digital switches instead of analog joysticks, but it does not explain how.

It would be even better if we had dedicated keycodes for this, EG `JS_UP`, `JS_DOWN`, `JS_LEFT`, and `JS_RIGHT`.

a-chol, 2021-06-26:
Hey, so in the current implementation, the buttons are just numbered and map are interpreted by the os in an os-specific way, I guess. You can define JOYSTICK_AXES_COUNT 0 to not have any axis defined and JOYSTICK_BUTTON_COUNT to something greater than 0 to only have buttons. Does that solve your use case?
I'm not sure if you can get semantic association for the buttons on the device side but I'll look it up.

skullydazed, 2021-06-26:
It feels like it doesn't, but maybe it does?

Let's start with a use case- Someone buys [a joystick and some arcade buttons](https://smile.amazon.com/JLF-TP-8YT-Joystick-OBSF-30-Buttons-Bundle/dp/B01FJFXFHK/) and connects them up to a promicro using directpins. Now they have 12 switches to assign keycodes to. The buttons are fairly obvious as you just assign `JS_BUTTON1`-`JS_BUTTON8`. What do I assign to the 4 switches in the joystick itself?

fauxpark, 2021-06-26:
How about these:
```
JS_PX
JS_NX
JS_PY
JS_NY
JS_PZ
JS_NZ
```

a-chol, 2021-06-26:
Ha! I guess I hadn't fully understood what you mean't in the initial request : a physical directional cross that sends analog values as axes not button presses?
You could implement it in your matrix_scan_*() function with custom keycodes and axes defined as virtual (see the example in the documentation). And I guess adding specific keycodes and support within qmk is easily doable as well. Is that what you meant?

a-chol, 2021-06-26:
Now that I look at it, I guess the sample in the doc is missing some code to write the actual value to the axis.

skullydazed, 2021-06-26:
> a physical directional cross that sends analog values as axes not button presses?

Is this what is happening when a commercial NES-style controller is plugged in? If so yes.

skullydazed, 2021-06-26:
For some more context, what I want in the end is for someone who knows little about how game controllers work to be able to use this for their custom controller. There's a lot of prior art here- fight sticks, flight controllers, console style controllers, etc. Right now you have to understand a lot of technical detail to make use of any of this, unless you stick solely to `JS_BUTTON<N>`.

fauxpark, 2021-06-27:
I suspect those NES controllers are using these for the D-pad, rather than sending fake analog values:
[image]

skullydazed, 2021-06-27:
I dug out an old [Logitech F510](https://smile.amazon.com/Logitech-Rumble-Gamepad-Support-Vibration/dp/B003VAM392/) I have to see what it does, and it does seem to supply max values as part of an axis. Using [this code](https://gist.github.com/jasonwhite/c5b2048c15993d285130) I got this from the cross axis on the left (which I can feel has switches, not pots):

```
Axis 2 at (     0, -32767)
Axis 2 at (     0,      0)
Axis 2 at (     0,  32767)
Axis 2 at (     0,      0)
Axis 2 at (-32767,      0)
Axis 2 at (     0,      0)
Axis 2 at ( 32767,      0)
Axis 2 at (     0,      0)
```

I get something similar when I plug in a gensis usb controller:

```
Axis 0 at (     0, -32767)
Axis 0 at (     0,      0)
Axis 0 at (     0,  32767)
Axis 0 at (     0,      0)
Axis 0 at (-32767,      0)
Axis 0 at (     0,      0)
Axis 0 at ( 32767,      0)
Axis 0 at (     0,      0)
```

And when I plug in a TG-16 controller:

```
Axis 2 at (     0, -32767)
Axis 2 at (     0,      0)
Axis 2 at (     0,  32767)
Axis 2 at (     0,      0)
Axis 2 at (-32767,      0)
Axis 2 at (     0,      0)
Axis 2 at ( 32767,      0)
Axis 2 at (     0,      0)
```

skullydazed, 2021-06-27:
On button assignments it would be useful to note standard button mappings. Just checking the three I have here there seem to be some commonalities.

TG-16 gamepad:

1 axis, 4 buttons

Crosspad controls Axis 2

| Gamepad Button | HID Button |
|-|-|
| Select | 8 |
| Run (Start) | 9 |
| II (left button) | 1 |
| I (right button) | 2 |

Genesis gamepad:

1 axis, 8 buttons

Crosspad controls Axis 0

| Gamepad Button | HID Button |
|-|-|
| Start | 9 |
| Mode (Right shoulder) | 8 |
| A | 1 |
| B | 2 |
| C | 4 |
| X | 0 |
| Y | 3 |
| Z | 5 |

Logitech F510, in primary mode<sup>1</sup>:

3 axis, 10 buttons

Left analog stick controls Axis 0<sup>2</sup>

Right analog stick controls Axis 1

Crosspad controls Axis 2

| Gamepad Button | HID Button |
|-|-|
| Back (Select) | 8 |
| Start | 9 |
| A | 1 |
| B | 2 |
| X | 0 |
| Y | 3 |
| LB | 

Logitech F510, in alternate mode<sup>1</sup>:

3 axis, 8 buttons

Left analog stick controls Axis 0<sup>3</sup>

Right analog stick left/right controls Axis 1. X values is always -32767 while Y value reflects position.

Right analog stick up/down controls Axis 2. X value reflects position while Y value is always -32767.

LT (left bottom trigger) controls Axis 1. X value reflects position while Y value is always 0. Resting position X value is -32767.

RT (right bottom trigger) controls Axis 2. X value is always -2 while Y value reflects position. Resting position Y value is -32767.

Crosspad is disabled unless Mode is pressed.<sup>3</sup>

| Gamepad Button | HID Button |
|-|-|
| Back (Select) | 8 |
| Start | 9 |
| A | 1 |
| B | 2 |
| X | 0 |
| Y | 3 |
| LB | 4 |
| RB | 5 |

Footnotes:
1. This controller can operate in two modes- primary and alternate. While in racing mode the crosspad is disabled and the LT and RT triggers switch from digital buttons to analog axis.
2. Pressing the "Mode" button causes the analog stick to act like the crosspad, reporting only maximum values to indicate direction.
3. Pressing the "Mode" button disables the left joystick and sends special values of `(0, -2)` and `(0,-779)` on Axis 0. This also enables the crosspad on Axis 0, but the Y value will always be -2 except when Up or Down are pressed.

davewongillies, 2021-12-14:
> For some more context, what I want in the end is for someone who knows little about how game controllers work to be able to use this for their custom controller. There's a lot of prior art here- [...]

I'm working on a project like this right now and either some documentation or predefined `JS_{UP,DOWN,LEFT,RIGHT}` keycodes would be super-helpful

jkutianski, 2022-02-03:
Some games/apps don't support D-PAD input.
I think the best bet here is define 14 new keycodes

For normal D-PAD (mapped to HID 0x90 Up, 0x91 Down, 0x92 Right, 0x93 Left)
```
DP_UP
DP_DOWN
DP_LEFT
DP_RIGHT
```

For D-PAD mapped to joystick
```
JS_UP
JS_DOWN
JS_LEFT
JS_RIGHT
JS_PRECISION
JS2_UP
JS2_DOWN
JS2_LEFT
JS2_RIGHT
JS2_PRECISION
```

This give us the freedom to create any combination possible (Especially if VIA/VIAL support this)

As example the [8Bitdo Lite](https://www.amazon.com/-/es/8Bitdo-Lite-inal%C3%A1mbrico-Bluetooth-Raspberry/dp/B08CRHQL17) has 2 D-PADs mapped to analog joysticks + 4 buttons used as D-PAD

closed 2022-07-13 by github-actions[bot]

davewongillies, 2024-03-19:
@5A52 I gave up and moved to the [GP2040-CE](https://gp2040-ce.info/) project instead. Its sort of like QMK, but specifically for joysticks and game controllers.

5A52, 2024-03-19:
@davewongillies thanks, RPI is a bit overkill for 5 button device. And I like QMK layers, just need exactly what was discussed here. Something like JS_UP JS_DOWN  JS_LEFT  JS_RGHT

davewongillies, 2024-03-19:
> @davewongillies thanks, RPI is a bit overkill for 5 button device. And I like QMK layers, just need exactly what was discussed here. Something like JS_UP JS_DOWN JS_LEFT JS_RGHT

Its a pipico, so its just a micro controller, so it not really overkill