qmk/qmk_firmware #13379: [Bug] LSFT_T(KC_LPRN) types 9 instead of (

stathissideris, 2021-06-29:
I have created a keymap ([see my fork](https://github.com/stathissideris/qmk_firmware/blob/master/keyboards/preonic/keymaps/stathissideris/keymap.c#L74)) where I'd like the right shift to type `(` when tapped. I have used `LSFT_T(KC_LPRN)` for this, but I get `9` instead.

## System Information

 - Keyboard: Preonic
   - rev3:  
 - Operating system: MacOS
 - AVR GCC version: avr-gcc (GCC) 5.4.0
 - ARM GCC version: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
 - QMK Firmware version: 0.13.15
 - Any keyboard related software installed?
   - [ ] AutoHotKey
   - [x] Karabiner - bug happens even when I quit Karabiner
   - [ ] Other:

fauxpark, 2021-06-29:
This is a limitation of how keycodes work in QMK: https://docs.qmk.fm/#/mod_tap?id=caveats

stathissideris, 2021-06-29:
Alright, thanks for the explanation! Is there a more low level way to achieve this, possibly in `process_record_user`?

sigprof, 2021-06-29:
You may try using the [Space Cadet](https://docs.qmk.fm/#/feature_space_cadet) feature (`KC_LSPO`, `KC_RSPC`) — it is even enabled by default. However, if you will be using other Mod-Tap or Layer-Tap keys in your keymap, Space Cadet can have some compatibility problems with them (the builtin tapping code does not treat Space Cadet keys as modifiers). And the behavior of the Space Cadet keys is different from Mod-Tap keys — the modifier is sent immediately when the key is pressed, and when the key is released without any other keypresses between the press and release, the tap keypress with the specified mods is sent. This may cause difficulties when trying to use Space Cadet with the Alt modifier, because the Alt tap will be visible to the OS.

Alternatively, you may try to use Mod-Tap keycodes with some basic keycodes, but replace the tap action using custom code inside `process_record_user()`:
```c
case LSFT_T(KC_9):
    if (record->tap.count > 0) {
        if (record->event.pressed) {
            register_code16(KC_LPRN);
        } else {
            unregister_code16(KC_LPRN);
        }
        return false;
    }
    return true;
```
This will effectively enable `IGNORE_MOD_TAP_INTERRUPT` for that key; if you don't like the resulting behavior, you may want to use `PERMISSIVE_HOLD`, or duplicate the behavior of the original Mod-Tap code in your `process_record_user()` implementation:
https://github.com/qmk/qmk_firmware/blob/5538bffb8eff5203fdabef8f4d8913999177fa53/tmk_core/common/action.c#L350-L361

stathissideris, 2021-06-29:
Many thanks for your comment, I'm new to QMK but I'll study the code you sent me and comment here about what worked for me for posterity.

stathissideris, 2021-06-29:
The solution srating with `case LSFT_T(KC_9):` worked perfectly for me, thanks.

closed 2021-06-29 by stathissideris