qmk/qmk_firmware #13431: [Bug] qmk setup seems to fail on WSL2

SirEelBiscuits, 2021-07-03:
## Describe the Bug

when running the command `qmk setup` when following the setup in readme, you get a failure from gcc

The failure can be safely ignored, and seems to relate to the fact you can't flash from WSL - I was able to build a new firmware despite the failure

Relevant part of the error message was:

```
gcc  -O2 -Wall `libusb-config --cflags` -c usbcalls.c -o usbcalls.o
In file included from usbcalls.c:19:
usb-libusb.c: In function ‘usbGetReport’:
usb-libusb.c:196:9: warning: value computed is not used [-Wunused-value]
  196 |         *len++;
      |         ^~~~~~
gcc  -O2 -Wall `libusb-config --cflags` -o bootloadHID main.o usbcalls.o `libusb-config --libs`
☒ Can't run `dfu-programmer --version`
```

## System Information

 - Operating system:
 - AVR GCC version: avr-gcc (GCC) 5.4.0
 - QMK Firmware version: 0.13.16
 - Any keyboard related software installed? No

fauxpark, 2021-07-03:
GCC did not fail here, it just gave a warning while compiling bootloadHID. What did fail is the check for `dfu-programmer --version`. What does that do on its own?

SirEelBiscuits, 2021-07-03:
I wasn't sure if the thing gcc was building was needed for the dfu-programmer,

Unless I'm mistaken, dfu programmer is a CLI tool for flashing microcontrollers and is known not to work in WSL - something the displayed text could have been a lot more clear about

fauxpark, 2021-07-03:
For the most part `qmk setup` treats WSL installs as normal Linux installs, as WSL(2) instances are essentially just Linux VMs. You are right though, it may be a good idea to skip checking certain things if WSL is detected, at least until proper USB support gets added...

SirEelBiscuits, 2021-07-03:
I can't remember when, but one of the scripts definitely already identifies WSL as special, and mentions needing QMK Toolkit to install firmware, but definitely if this is an expected situation, it should not show as a failure, and should instead provide the information a user needs to carry on

ReVanTis, 2021-07-05:
Try export these variables in your .bashrc, with your windows versions (.exe):
```
    export DFU_PROGRAMMER=/mnt/c/qmk_path/dfu-programmer.exe
    export DFU_UTIL=/mnt/c/qmk_path/dfu-util.exe
    export TEENSY_LOADER_CLI=/mnt/c/qmk_path/teensy_loader_cli.exe
    export BATCHISP=/mnt/c/qmk_path/batchisp.exe
```

closed 2022-08-28 by zvecr