qmk/qmk_firmware #13446: Does QMK support column output using a shift register?

j4cobgarby, 2021-07-04:
I designed a keyboard PCB where the column output is driven with a shift register, so that more columns can be achieved using a lot less microcontroller pins. 

I was planning on writing the firmware for it from scratch (or using the LUFA library, more likely, but I'm having issues with getting that to work...), and came across QMK, which looks great. I'm not sure exactly how to create a new keyboard, but it seems I have to specify a microcontroller pin for each column and row, which is not ideal for me. 

Is there a way around this?

tzarc, 2021-07-04:
Which shift register? 74xx595?

j4cobgarby, 2021-07-05:
> Which shift register? 74xx595?

Yep, that's the one, specifically a 74HC595

j4cobgarby, 2021-07-05:
Just found this (https://beta.docs.qmk.fm/using-qmk/hardware-features/custom_matrix) page which looks like what I need.

tzarc, 2021-07-05:
I've got an implementation using custom matrix here:
https://github.com/tzarc/qmk_build/blob/master/tzarc-cyclone/shiftreg_matrix.c

It's using '595s as row drivers, and '165s as column inputs. Should be documented enough for you to pull out what you need.

j4cobgarby, 2021-07-05:
That looks great, thanks!

closed 2021-07-05 by j4cobgarby