qmk/qmk_firmware #13458: [Feature Request]

pears0, 2021-07-05:
### Description

Add a way to re-point QMK too an avr-gcc toolchain that's not installed ont eh default path

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ x] New behavior

## Description

Lots of OS's are now moving on from avr-gcc 8.x but the newer versions create much larger file sizes as you are aware. Adding an option to tell qmk to use a differnt toolchain so as not to conflict with other build environments and allow people to use other toolchains other than what's installed by their OSes would be extreamly useful, even more so now with the issues 10.x and 11.x are throwing up with compile sizes.

0mp, 2021-08-13:
Wouldn't setting PATH be sufficient?

0mp, 2021-08-17:
Another option could be to create a flie in the project root directory called "makefile" with the following contents:

```
include Makefile
PATH:=	/new:${PATH}
```

then, when you execute `gmake` (or just `make` depending on your platform) everything should just work as you want it to.

closed 2022-04-17 by stale[bot]