qmk/qmk_firmware #13463: [Bug] Qmk setup error

Samaine, 2021-07-06:
When running the command `qmk compile -kb gmmk/pro -km default`

returns with

`QMK Firmware 0.13.18
Making gmmk/pro with keymap default

arm-none-eabi-gcc.exe (GCC) 10.1.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Linking: .build/gmmk_pro_default.elf
                            [ERRORS]
 |
 | arm-none-eabi-gcc.exe: error: .build/obj_gmmk_pro_default/lib/printf/printf.o: No such file or directory
 |
make[1]: *** [tmk_core/rules.mk:344: .build/gmmk_pro_default.elf] error 1
Make finished with errors
make: *** [Makefile:530: gmmk/pro:default] error 1`

fauxpark, 2021-07-06:
Run `qmk doctor`.

Samaine, 2021-07-06:
```Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.10
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.```

Samaine, 2021-07-06:
> Run `qmk doctor`.

```
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.10
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.```

laukit1997, 2021-07-08:
i have the same problem, same error occurs on everytime i try to compile

closed 2022-01-09 by stale[bot]