qmk/qmk_firmware #13469: [Bug] With Autoshift Enabled, holding Enter does not send Shift+Enter

VSantillo, 2021-07-06:
## Describe the Bug

With Autoshift Enabled, holding Enter does not send Shift + Enter.

## System Information

 - Keyboard: DZ60 ANSI 
   - Revision (if applicable): v2  
 - Operating system: MacOS
 - AVR GCC version: 
`avr-gcc (Homebrew AVR GCC 8.4.0_1) 8.4.0`
 - ARM GCC version: 
`arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]`
 - QMK Firmware version:
`0.13.2`
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

When Tab is held, it is treated as a Shift + Tab. My editor has a shortcut for Shift + Tab and Shift + Enter, but I had to program in a tap dance for Shift + Enter. Not the end of the world, but I think it's inconsistent.

zvecr, 2021-07-06:
Right now its the expected behaviour, but can be overridden at the keymap level by providing your own `get_auto_shifted_key` implementation.

<https://docs.qmk.fm/#/feature_auto_shift?id=auto-shift-per-key>

VSantillo, 2021-07-06:
Thanks! When I had read it earlier, I thought that the omission was unintentional since there was no explanation why.

arlaneenalra, 2023-05-19:
I actually came here looking for a way to pull tab out of the autoshift list while keeping the symbol characters present (for example `][=_}{` ). Would it make sense to do split the `NO_AUTO_SHIFT_SPECIAL` macro up to handle the symbol characters and tab/enter characters seperately?