qmk/qmk_firmware #13474: [Bug] Cannot Compile, cywin_exception.

jonspraggins, 2021-07-06:
## Describe the Bug
Attempting to compile for MassDrop CTRL keyboard.
```
[Jon@Jon-MBBC ~]$ qmk compile -kb massdrop/ctrl -km default
Ψ Compiling keymap with make -j 1 massdrop/ctrl:default

QMK Firmware 0.13.19
      0 [] make 2484 cygwin_exception::open_stackdumpfile: Dumping stack trace to make.exe.stackdump
☒ Invalid return_code: 35584
```

## System Information

 - Keyboard:  MassDrop CTRL
   - Revision (if applicable):  
 - Operating system: Windows 10
 - AVR GCC version: avr-gcc.exe (GCC) 8.4.0

 - ARM GCC version: arm-none-eabi-gcc.exe (GCC) 10.1.0

 - QMK Firmware version: 0.13.19

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

`[Jon@Jon-MBBC ~]$ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 0.1.0
Ψ QMK home: C:/Users/Jon/qmk_firmware
Ψ Detected Windows 10 (10.0.19041).
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.10
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.`

reversefold, 2021-07-06:
Having the same problem myself.
```
[paper@CORSAIRONE qmk_firmware]$ qmk compile -kb ergodox_ez/glow -km default
Ψ Compiling keymap with make -j 1 ergodox_ez/glow:default

QMK Firmware 0.13.19
      1 [] make 680 cygwin_exception::open_stackdumpfile: Dumping stack trace to make.exe.stackdump
☒ Invalid return_code: 35584
```

```
[paper@CORSAIRONE qmk_firmware]$ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 0.1.0
Ψ QMK home: C:/Users/paper/qmk_firmware
Ψ Detected Windows 10 (10.0.19041).
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.10
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go
```

Samaine, 2021-07-06:
Can confirm. Experiencing same issues.
```
Ψ Compiling keymap with make -j 1 gmmk/pro:default

QMK Firmware 0.13.18
      0 [] make 639 cygwin_exception::open_stackdumpfile: Dumping stack trace to make.exe.stackdump
☒ Invalid return_code: 35584
```

maxronner, 2021-07-06:
Also affected by this issue.

```
Ψ Compiling keymap with make -j 1 planck/rev6:default

QMK Firmware 0.13.12
      0 [] make 1945 cygwin_exception::open_stackdumpfile: Dumping stack trace to make.exe.stackdump
☒ Invalid return_code: 35584
```
```
Ψ CLI version: 0.1.0
Ψ QMK home: C:/Users/myuser/qmk_firmware
Ψ Detected Windows 10 (10.0.19041).
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 10.1.0
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.10
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go
```

tzarc, 2021-07-06:
Dupe of  #13416.

closed 2021-07-06 by tzarc