qmk/qmk_firmware #13478: [Bug] Compile error for Proton C with JOYSTICK_ENABLE

nyitraig, 2021-07-06:
## Describe the Bug

I was trying to add an [analog joystick](https://www.adafruit.com/product/512) to a ProtonC following [this](https://beta.docs.qmk.fm/using-qmk/hardware-features/feature_joystick) guide.

I made the following modifications:
```c
// rules.mk
JOYSTICK_ENABLE = analog
```
```c
// config.h
#define JOYSTICK_BUTTON_COUNT 0
#define JOYSTICK_AXES_COUNT 2
```
```c
// keymap.h
#include "joystick.h"

#ifndef ADC_PIN
#    define ADC_PIN1 F6
#    define ADC_PIN2 F7
#endif

joystick_config_t joystick_axes[JOYSTICK_AXES_COUNT] = {
    [0] = JOYSTICK_AXIS_IN(ADC_PIN1, 900, 575, 285),
    [1] = JOYSTICK_AXIS_IN(ADC_PIN2, 900, 575, 285),
};
```

During compile I get the following error:
```
Compiling: quantum/process_keycode/process_joystick.c                                               [OK]
Compiling: quantum/joystick.c                                                                       [OK]
Compiling: drivers/chibios/analog.c                                                                drivers/chibios/analog.c:23:6: error: #error "You need to set HAL_USE_ADC to TRUE in your halconf.h to use the ADC."
   23 | #    error "You need to set HAL_USE_ADC to TRUE in your halconf.h to use the ADC."
      |      ^~~~~
drivers/chibios/analog.c:27:6: error: #error "You need to set one of the 'STM32_ADC_USE_ADCx' settings to TRUE in your mcuconf.h to use the ADC."
   27 | #    error "You need to set one of the 'STM32_ADC_USE_ADCx' settings to TRUE in your mcuconf.h to use the ADC."
      |      ^~~~~
drivers/chibios/analog.c:111:8: error: unknown type name 'ADCConfig'
  111 | static ADCConfig   adcCfg = {};
      |        ^~~~~~~~~

...
```

## System Information

 - Keyboard:  QMK Proton C
   - Revision (if applicable):  Rev. 1
 - Operating system: Windows 10 Pro N
 - AVR GCC version: 8.4.0
 - ARM GCC version: 10.1.0
 - QMK Firmware version: 0.13.17

## Additional Context

I noticed there is a handwired example with a joystick under [`keyboards/handwired/onekey`](https://github.com/qmk/qmk_firmware/tree/master/keyboards/handwired/onekey), I tried building it with the following results:
* `$ make handwired/onekey/proton_c:joystick` same error as above
* `$ make handwired/onekey/promicro:joystick` successful build
* `$ make handwired/onekey/promicro:joystick CTPC=yes` same error as above

drashna, 2021-07-14:
The errors are telling you want you need to do to fix this. 

> You need to set HAL_USE_ADC to TRUE in your halconf.h to use the ADC
> You need to set one of the 'STM32_ADC_USE_ADCx' settings to TRUE in your mcuconf.h to use the ADC 

Basically, you need to add these files to enable analog pin support. 

You can see an example of how to do this here: 
https://github.com/drashna/qmk_firmware/tree/877e81f9633d1a1f73e741a04f08cf2fb9522974/layouts/community/split_3x6_3/drashna

nyitraig, 2021-07-16:
Thank you so so much! It worked!

drashna, 2021-08-09:
I'm glad to hear it!

closed 2021-08-09 by drashna