qmk/qmk_firmware #13483: [Bug] Weird behavior with RGBLIGHT_LAYER_SEGMENTS

LechuckThePirate, 2021-07-07:
Hi everyone.

I have my keyboard configured to use different colors when different layers are enabled. Following the instructions in QMK site I found this code and it works perfectly:

```
const rgblight_segment_t PROGMEM my_qwerty_layer[]   = RGBLIGHT_LAYER_SEGMENTS({0, RGBLED_NUM, HSV_CYAN});
const rgblight_segment_t PROGMEM my_raise_layer[]    = RGBLIGHT_LAYER_SEGMENTS({0, RGBLED_NUM, HSV_BLUE});
const rgblight_segment_t PROGMEM my_lower_layer[]    = RGBLIGHT_LAYER_SEGMENTS({0, RGBLED_NUM, HSV_ORANGE});
const rgblight_segment_t PROGMEM my_adjust_layer[]   = RGBLIGHT_LAYER_SEGMENTS({0, RGBLED_NUM, HSV_RED});
const rgblight_segment_t PROGMEM my_capslock_layer[] = RGBLIGHT_LAYER_SEGMENTS({0, RGBLED_NUM, HSV_GREEN});

// Now define the array of layers. Later layers take precedence
const rgblight_segment_t* const PROGMEM my_rgb_layers[] =
RGBLIGHT_LAYERS_LIST(
            my_qwerty_layer,
            my_raise_layer,
            my_lower_layer,
            my_adjust_layer,
            my_capslock_layer
        );
void keyboard_post_init_user(void) {
    rgblight_layers = my_rgb_layers;
}
// Acitvate the default rgb layer when the _QWERTY layer is enabled
layer_state_t default_layer_state_set_user(layer_state_t state) {
rgblight_set_layer_state(0, layer_state_cmp(state, _QWERTY));
return state;
}
// Activate rgb layer for caps when capslock is enabled
bool led_update_user(led_t led_state) {
rgblight_set_layer_state(4, led_state.caps_lock);
return true;
}
// Activate the rgb layer according to the active keyboard layer
layer_state_t layer_state_set_user(layer_state_t state) {
rgblight_set_layer_state(1, layer_state_cmp(state, _RAISE));
rgblight_set_layer_state(2, layer_state_cmp(state, _LOWER));
rgblight_set_layer_state(3, layer_state_cmp(state, _ADJUST));
return state;
}
```

This made my keyoard had a default uniform lighting in cyan, then change to orange, blue, red, etc... when switching layers.

Today I received a new set of keys in Pom Jelly, and I wanted to change the layout to white normal keys and cyan special keys... so I did this changes according with my PCB board rgb lights numbering...

```
const rgblight_segment_t PROGMEM my_qwerty_layer[]   =
RGBLIGHT_LAYER_SEGMENTS(
    {0, 6, HSV_WHITE},
    {6, 1, HSV_CYAN},
    {7, 6, HSV_WHITE},
    {13, 2, HSV_CYAN},
    {15, 9, HSV_WHITE},
    {24, 3, HSV_CYAN}
    );
```

After this change everything looked good, but when I flashed the second half of the keyboard (it's a corne split), the right part started behaving weird, sometimes, turning all rgb to red on it's own, and then sometimes having multiple resets in a loop rendering the keyboard unusable.

I thought maybe I was doing something wrong with my new changes, so I just reverted the changes, by just changing the HSV_CYAN color of the qwerty (main) layer to HSV_WHITE like this:

`const rgblight_segment_t PROGMEM my_qwerty_layer[]   = RGBLIGHT_LAYER_SEGMENTS({0, RGBLED_NUM, HSV_WHITE});
`
To my surprise, the weird behavior was still there. The right part of the keyboard turning red and occasional resets...

Then, just to know that I didn't break any hardware part, I changed the white color to the original cyan, to have it like before any change... AND IT WORKS!! so, it seems if I set it to cyan everything works as expected, but if I set it to white (I tried with HSV_AZURE as well, it looks white though) then it starts doing the weird color changes and resets again...

Anyone has the slightest idea of what is the problem? It's totally nuts ... the only difference in the code is the color I'm using!! Cyan works as a charm, White makes everything go crazy :(

## System Information

 - Keyboard: Corne (crkbd)
   - Revision (if applicable): rev1  
 - Operating system: Windows 10
 - AVR GCC version:  avr-gcc.exe (GCC) 8.4.0

 - ARM GCC version: arm-none-eabi-gcc.exe (GCC) 10.1.0

 - QMK Firmware version: 0.13.2

 - Any keyboard related software installed? 
   - [ NO] AutoHotKey
   - [ NO] Karabiner
   - [] Other:

daskygit, 2021-07-07:
Have you limited the brightness? White draws a good deal more current, the 500mA polyfuse will be tripped at maximum brightness with white very quickly.

`#define RGBLIGHT_LIMIT_VAL 120` is probably about the maximum it'll manage.

LechuckThePirate, 2021-07-07:
> Have you limited the brightness? White draws a good deal more current, the 500mA polyfuse will be tripped at maximum brightness with white very quickly.
> 
> `#define RGBLIGHT_LIMIT_VAL 120` is probably about the maximum it'll manage.

I think you're right... I switched to 120 and it doesn't seem to happen anymore ... but yeah.. it looks a little dimmed for my taste... 

I'll play to check ho much can I raise it to without going into trouble.

Is it safe to play with that value? Or can I break anything?

daskygit, 2021-07-07:
I can't imagine you'll get much more than that all white at least, maybe 150ish. If you have a usb power meter it'd be easier to fine tune it. I can't remember the exact value I settled on. 

You can play with it a little, it's not the best idea to keep tripping the polyfuse.

LechuckThePirate, 2021-07-07:
It seems to be safe up to 150 at least... it's ok... so, no bug... just my hardware doing strange things.. haha.

@daskygit do you know if there is any way to increase the capability of the hardware to raise the brightness?? changing any of the components?

daskygit, 2021-07-07:
You could replace the polyfuse and diode but you run the risk of overloading the upstream port. 

I'm actually using a replacement controller on my corne that has limited usb pd support but allows for up to 2.5A. I have plans on releasing the next revision on github.

LechuckThePirate, 2021-07-07:
Ok thanks for the help!! It worked now for hours without any trouble. I'm closing the issue since it's not related to QMK.

closed 2021-07-07 by LechuckThePirate