qmk/qmk_firmware #13485: [Bug] Unable to use qmk with hasu converter

nosleduc, 2021-07-07:
I have bought a hasu converter and I'm trying to use qmk on it.
It was delivered with a tmk firmware that worked out of the box.
I tested and compiled a keymap of my own with the tmk firmware and it works perfectly.

Now I'm trying to use qmk instead to get tap dance.
I have installed the requirements to compile a keymap.
The compilation succeeded and the board was flashed but nothing works after.

I tried to use the QMK Toolbox instead. I downloaded the keyboard converter/usb_usb/hasu from the toolbox.
It flashes correctly the board but nothing works after.

I'm using the latest version of the master branch on macos mojave and the revision G for the hasu converter.
I have followed the readme

Is there something special to do after the flashing ?

CruelTott, 2021-07-11:
Have you tried more than one keyboard with it?

I recently got a converter and noticed that my logitech keyboard didn't work with qmk, but worked fine with tmk.

I found that Hasu had recently done some updates to his firmware to increase compatibility.
https://geekhack.org/index.php?topic=69169.msg2996293#msg2996293

I found that applying the fixes at the bottom of the issue linked from that forum post or updating the entire USB_Host_Shield_2.0 folder under qmk_firmware/tmk_core/protocol/usb_hid fixed it for me.

nosleduc, 2021-07-11:
I have tried with two keyboards logitech ergo k860 and iqunix f96. It worked out of the box with the tmk firmware.

These keyboards do not work with qmk. I don't think it comes from the keyboards. After flashing the board with the qmk firmware, the qmk toolbox does not see the converter. With the tmk firmware, the qmk toolbox sees the converter without any keyboards attached to it.

I tried what you said I have updated USB_Host_Shield_2.0 with the contents of the one in the tmk repository it does not change anything

iwbrhwfy-source, 2021-07-16:
try pro_micro and hasu becasue it differnet cpu clock

CruelTott, 2021-07-16:
This is probably not going to help,
but I tried flashing tmk onto mine and then flashing qmk using the default from qmk-toolbox to see if I had forgotten if I did any extra step the first time to get it working.

I happened to have it plugged into an unpowered usb hub which caused it to work with tmk but not qmk.

Turns out qmk advertises a higher power consumption requirement than tmk.
If I defined USB_MAX_POWER_CONSUMPTION to 100 it worked in that usb port too.

It worked without changes in basically any other usb-port, so it seems unlikely that that would be the problem for you.
But that is at least one more difference between tmk and qmk.

nosleduc, 2021-07-16:
Thanks for your help. I have put USB_MAX_POWER_CONSUMPTION to 100.

I bought the hasu to use it on several keyboards to have the same behaviour.

My first  idea was to use it mainly with the dygma raise. This one does not use qmk and it lacks several features. But the dygma raise does not work with tmk and qmk.

Then I tried with with iqunix f96 supported by tmk but not with qmk.

I borrowed two keyboards from friends. 
I tested a moonlander. It does not make sense to add the hasu on it but useful to test it. The hasu works with qmk and with the moonlander. I have also tested a logitech mx keys and it also worked.

I need to test again the logitech ergo.
It seems that the dygma raise does not use a standard boot.

The iqunix f96 is a nice keyboard with an excellent quality build but with a poor firmware. 
It will be great to support it.

closed 2021-07-16 by nosleduc

rubenderaeve, 2021-12-24:
> I have tried with two keyboards logitech ergo k860 and iqunix f96. It worked out of the box with the tmk firmware.
>
> These keyboards do not work with qmk. I don't think it comes from the keyboards. [...]

> I have tried with two keyboards logitech ergo k860 and iqunix f96. It worked out of the box with the tmk firmware.
>
> These keyboards do not work with qmk. I don't think it comes from the keyboards. [...]

@nosleduc  Did you managed to rum the qmk firmware on the logitech ergo k860 ?

nosleduc, 2021-12-25:
With the latest version of [https://config.qmk.fm/#/converter/usb_usb/hasu/LAYOUT_all](url) it works