qmk/qmk_firmware #13492: Add support for lighting layers for RGB matrix

mike1808, 2021-07-08:
"Lighting layers" is a feature allowing to set the color of individual LEDs depending on the active layer. Currently, the lighting layers are only supported for [RGB lighting](https://docs.qmk.fm/#/feature_rgblight?id=lighting-layers).

## Feature Request Type

- [ ] Core functionality
- [x] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

I imagine lighting layers for RGB matrix should be defined in a similar way as key layouts. For example, if have the following layout for our keyboard:

```c
// A B C 
// D E F
// G H K
# define LAYOUT( \
    k00, k01, k02, \
    k03, k04, k05, \
    k06, k09, k10
) \
{ \
   { k00, k01, k02 }, \
   { k03, k04, k05 }, \
   { k06,  k07, k08 } \
}
```
It's RGB matrix lighting layers can be defined in the following way:
```c
const uint8_t ledmap PROGMEM ledmaps[][DRIVER_LED_TOTAL][3]= {
         [_BASE] = LEDMAP(
                         ____, ____, ____,
                         ____, ____, ____,
                         ____, ____, ____
        )
        [_FN] = LEDMAP(
                         HSV_RED, ____, ____,
                         ____, ____, HSV_RED,
                         ____, HSV_BLUE, ____
        )
};
```

Where `LEDMAP` is a macro mapping every key on the layout to the appropriate LED index, like `g_led_config`.

I have some implementation of this feature in my [keymap](https://github.com/qmk/qmk_firmware/pull/13398/files#diff-cbdb7f7bb6bf143b67fe039b17fab7138e841391edf401b77adaa2db5f8133b8R83) #13398

Looking forward for comments and suggestions

mike1808, 2021-07-08:
As mentioned by `trankillity` in the QMK discord, by having `layer_num` * `DRIVER_LED_TOTAL` * `3` can add a significant size to the firmware we can think about some ways to compact the definition of the RGB layout. One of the suggestions is to drop the `value` of the HSV color and shave 33%.

mike1808, 2021-07-08:
Another consideration is that not all keys might have LED index and also `DRIVER_LED_TOTAL` might be larger than the number of keys, e.g. there might be indicator and underglow LEDs. I was thinking to that that can be mitigated by just dropping them in the `LEDMAP` macro. Or instead of dropping, we can add them in that macro, for example, we can have underglow LEDs defined near the layout. 

```c
const uint8_t ledmap PROGMEM ledmaps[][DRIVER_LED_TOTAL][3]= {
         [_BASE] = LEDMAP(
               HSV_RED,     ____, ____, ____,    HSV_RED
                                      ____, ____, ____,
                                      ____, ____, ____
        )
        [_FN] = LEDMAP(
          HSV_BLUE      HSV_RED, ____, ____,      HSV_BLUE
                                  ____, ____, HSV_RED,
                                  ____, HSV_BLUE, ____
        )
};

trankillity, 2021-07-08:
Here's a more complicated, real world example of the LED mapping on a Corne V3 PCB. The indexes in this are not even close to the physical mapping, so this functionality was definitely required for me.
```c
#define LEDLAYOUT_split_3x6_3(\
  L25, L24, L19, L18, L11, L10,           R10, R11, R18, R19, R24, R25, \
  L26, L23, L20, L17, L12, L09,           R09, R12, R17, R20, R23, R26, \
  L27, L22, L21, L16, L13, L08,           R08, R13, R16, R21, R22, R27, \
                      L15, L14, L07, R07, R14, R15, \
       L03,      L02,      L01,           R01,      R02,      R03, \
       L04,      L05,      L06,           R06,      R05,      R04 \
) {{L01}, {L02}, {L03}, {L04}, {L05}, {L06}, {L07}, {L08}, {L09}, {L10}, {L11}, {L12}, {L13}, {L14}, {L15}, {L16}, {L17}, {L18}, {L19}, {L20}, {L21}, {L22}, {L23}, {L24}, {L25}, {L26}, {L27}, \
  {R01}, {R02}, {R03}, {R04}, {R05}, {R06}, {R07}, {R08}, {R09}, {R10}, {R11}, {R12}, {R13}, {R14}, {R15}, {R16}, {R17}, {R18}, {R19}, {R20}, {R21}, {R22}, {R23}, {R24}, {R25}, {R26}, {R27}}
```
Note that the map also includes the underglow LEDs, which are skipped in the Corne's `g_led_config`.

drashna, 2021-07-11:
Instead of `uint8_t`, I'd rather this use something like 
```c
struct {
   HSV hsv;
   bool overwrite_led;
} ledmap_t;

const ledmap_t PROGMEM ledmap[][DRIVER_LED_TOTAL] = {
```
This way, each location can be set to determine if the LED at that location should be overwritten or not.

Myridium, 2021-09-22:
Would be nice to have this feature. Does this affect only some RGB keyboards or all? The Moonlander seems to use RGB Matrix at least.