qmk/qmk_firmware #13524: [Feature Request] GMMK Pro ISO support

Hatbuster, 2021-07-12:
Now that the GMMK Pro ISO hit the market, I'd like it to be supported by QMK, too

## Feature Request Type

- [X] Core functionality
- [X] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

The GMMK Pro ISO PCB works very similarly to the ANSI one.
Going by the matrix currently available for the ANSI version, Backslash is ka2 and no longer needed, since it does not exist on the ISO board.
The new keys needed are k23 (KC_NUBS) and k84(KC_NUHS).

Copying the proposed AW20216S RGB implementation into my experimental ISO firmware results in good behavior, except for NUBS and NUHS not being lit. Have not tried the competing one.

I have some codework on my end, but I am very inexperienced with the project and github, so I'd love if someone more experienced could do what glorious should have done 2 weeks ago.

cheers.

tipok, 2021-07-13:
I've created a PR adding an ISO layout #13532

Hatbuster, 2021-07-13:
Thanks for getting that comit in!

I wasn't sure if a new keyboard needed to be created since it's a new PCB or if a new layout sufficed. Mostly because I don't have any clue about how the RGB configuration works.

We'll probably have to see about RGB once the Glorious PR gets merged or rejected. RGB is secondary anyways, at least to me.

Gigahawk, 2021-07-16:
If someone with an ISO PCB and a multimeter can probe the dark LEDs to figure out which rows/columns they belong to it shouldn't be that hard to get those LEDs lit up.

[image]

I just need to know:
- Continuity between anode pin of a dark LED to one that's already lit
    - Anode is the pin marked with the `+` on the PCB.
    - All the anode pins should be connected across one `SWxx_x` column in the table above
- Continuity between a cathode pin of a dark LED to the same cathode pin on one that's already lit
    - Cathode pins are any of the other three pins, just make sure to pick the same corner (relative to the `+` mark on the PCB)
    - All the cathode pins should be connected across one `CSxx_x` row in the table above

Presumably the ISO specific keys correspond to one of the `Kxx` entries in the table above.

Giesd, 2021-07-16:
I just measured it:

LED 83 (that's the label on the PCB) belongs to the key to the right of Left Shift (<> in ISO-DE, \ in ISO-UK) and is mapped to K45, so SW1_1 and CS4_1 - CS6_1.
LED43 is for the # key and is mapped to K42, so SW10_1 and CS1_1 - CS3_1.

I didn't test it yet but maybe it helps.

Gigahawk, 2021-07-20:
Fix provided, this can probably be closed now.

Hatbuster, 2021-07-20:
Key- and RGB matrix is in.

Cheers guys. Thanks for stepping in where glorious should have.

PR #13570 (2021-07-20): Split GMMK Pro PCBs into separate revisions, fix RGB mapping on ISO

## Description

RGB mapping is unique between ISO and ANSI GMMK Pro PCBs, split each version into a separate revision for proper support.

@Hatbuster @Giesd @tipok I don't actually have an ISO model to test, on so if you could please confirm this actually works.

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [x] Keyboard (addition or update)
- [x] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #13524

## Checklis [... cut]
closed 2021-07-20 by Hatbuster