qmk/qmk_firmware #13534: [Feature Request] Hardware support for HC-05 boards flashed with RN-42 firmware.

Leonezz, 2021-07-13:
The HC-05 boards are in the [Bluetooth doc page](https://docs.qmk.fm/#/feature_bluetooth)'s `Not Supported Yet but possible` list, is it in development?

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description
I'd really like to see HC-05 bluetooth boards support in QMK, it's really a good choice for keyboard bluetooth function. It is a lot cheaper and easier to buy on the Taobao.com than the RN-42 and AdafruitBLE boards.

I did some code work on my end, I tried to operate UART to drive the HC-05 boards. but I am not familiar with the QMK APIs and the HID message frame structure. As the board is on the `Not Support Yet but possible` list, maybe someone are working on this?

fauxpark, 2021-07-13:
RN42 modules are supported in QMK, however they are pretty old at this point (they're Bluetooth 2.1) so it's likely that support will actually be dropped soon.

Leonezz, 2021-07-14:
> RN42 modules are supported in QMK, however they are pretty old at this point (they're Bluetooth 2.1) so it's likely that support will actually be dropped soon.

Will new bluetooth modules be supported soon? like HC-05 modules.

fauxpark, 2021-07-14:
HC-05/RN-42 modules aren't new at all - as I said, they're Bluetooth 2.1).
Due to licensing issues QMK cannot run directly on nRF MCUs, so the only option for QMK is external modules that communicate over something like SPI, I2C, UART etc.
At the moment this essentially means RN-42/HC-05, or the Adafruit Bluefruit LE, either the SPI Friend module or the Feather 32U4:
https://www.adafruit.com/product/2633
https://www.adafruit.com/product/2829
The problem is that some features of QMK (eg. NKRO) are not available because the module itself controls the HID descriptors.

If you are really set on Bluetooth, I would recommend you look into ZMK instead.

Leonezz, 2021-07-14:
That makes sense, some bluetooth modules are limited for licensing issues and some control the HID itself. 
Now I understand the problems, thanks a lot.

closed 2021-07-14 by Leonezz