qmk/qmk_firmware #13535: [Bug] qmk flash hangs on OSX (Big Sur)

ahri, 2021-07-13:
## Describe the Bug

I followed the instructions to compile and flash durgod/k320; this worked fine on Linux. I then tried to get it working on OSX and have compiled, but cannot flash. This is true both for the command line "qmk flash" and for the QMK Toolbox.

It appears to hang when trying to erase. I wonder if I missed a step to provide adequate permissions?

Output from QMK Toolkit:

```
*** STM32 DFU device connected: STMicroelectronics STM32  BOOTLOADER (0483:DF11:2200)
*** Attempting to flash, please don't remove device
>>> dfu-util -a 0 -d 0483:DF11 -s 0x8000000:leave -D /Users/adam.piper/repos/qmk_firmware/durgod_k320_ahri.bin
    dfu-util 0.9
    
    Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
    Copyright 2010-2016 Tormod Volden and Stefan Schmidt
    This program is Free Software and has ABSOLUTELY NO WARRANTY
    Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
    
    Opening DFU capable USB device...
    ID 0483:df11
    Run-time device DFU version 011a
    Claiming USB DFU Interface...
    Setting Alternate Setting #0 ...
    Determining device status: state = dfuERROR, status = 10
    dfuERROR, clearing status
    Determining device status: state = dfuIDLE, status = 0
    dfuIDLE, continuing
    DFU mode device DFU version 011a
    Device returned transfer size 2048
    DfuSe interface name: "Internal Flash  "
    Downloading to address = 0x08000000, size = 21332
    
Download	[                         ]   0%            0 bytesdfu-util: Error during special command "ERASE_PAGE" get_status
*** STM32 DFU device disconnected: STMicroelectronics STM32  BOOTLOADER (0483:DF11:2200)
```

Output from CLI:

```
% qmk flash
Ψ Compiling keymap with make -j 1 durgod/k320:ahri:flash

QMK Firmware 0.13.19
Making durgod/k320 with keymap ahri and target flash

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  21332	      0	  21332	   5354	.build/durgod_k320_ahri.hex

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  21332	      0	  21332	   5354	.build/durgod_k320_ahri.hex

Copying durgod_k320_ahri.bin to qmk_firmware folder                                                 [OK]
Bootloader not found. Trying again every 0.5s (Ctrl+C to cancel)............
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading element to address = 0x08000000, size = 21332
Erase   	[                         ]   0%            0 bytes
```
(it just sits at 0 bytes seemingly forever - I left it for 20 mins the first time as I was scared of bricking the keyboard!)

In both cases after unplugging and reconnecting the keyboard everything is fine, so it's not erasing anything.

## System Information

 - Keyboard:  Durgod K320
 - Operating system: OSX Big Sur
 - AVR GCC version: avr-gcc (Homebrew AVR GCC 8.4.0_2) 8.4.0
 - ARM GCC version: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
 - QMK Firmware version: 0.13.19

masoudl, 2021-07-14:
Any luck? I am having the same issue.

ahri, 2021-07-14:
@masoudl none yet! I did start constructing a Docker container to cheat by running qmk in linux in osx, but passing the USB device all the way through seems like a pain, and without knowing what the problem is it may not even work.

ahri, 2021-07-14:
I tried installing VirtualBox on my OSX host, and installing Ubuntu in there, following the Linux steps, making sure the USB device is passed through (which it isn't by default, when in bootloader mode) and then doing `qmk flash` - no dice.

It behaves exactly as outlined above, just hangs trying to erase, and when pulled out and reconnected the keyboard works fine.

So yeah, when trying to flash on an OSX (Bug Sur) host writing just hangs with no feedback as to why.