qmk/qmk_firmware #13549: KBD67 Lite R1 flashing issues in QMK

mahn123, 2021-07-15:
Hi everyone! Im kind of new to the keyboard community but Im having some problems with my KBD67 Lite R1. Ever since plugging the board into my new pc, the keys would not register. I ended up flashing the board through qmk with the correct via files. This worked for about a week as I had to reflash the board everytime I turned on my computer. Only until today did reflashing stopped working. Here are some of the logs from qmk. 

Atmel DFU device connected (libusb0): Atmel Corp. ATmega32U4 (03EB:2FF4:0000)
>>> dfu-programmer.exe atmega32u4 erase --force
>>> dfu-programmer.exe atmega32u4 flash --force --eeprom "reset.eep"

Please reflash device with firmware now
    Erasing flash...  Success
    0%                            100%  Programming 0x80 bytes...
    Checking memory from 0x0 to 0x6FFF...  Not blank at 0x1.
    [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
    0%                            100%  Reading 0x400 bytes...
    [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
    Validating...  0 invalid bytes in program region, 896 outside region.
    FAIL
    Memory did not validate. Did you erase?

Attempting to flash, please don't remove device
>>> dfu-programmer.exe atmega32u4 erase --force
    Erasing flash...  Success
    Checking memory from 0x0 to 0x6FFF...  Not blank at 0x1.
>>> dfu-programmer.exe atmega32u4 flash --force "C:\Users\Matthew\Downloads\kbdfans_kbd67_mkiirgb_v2_via (1).hex"
    0%                            100%  Programming 0x5B00 bytes...
>>> dfu-programmer.exe atmega32u4 reset
    [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
    0%                            100%  Reading 0x7000 bytes...
    [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
    Validating...  ERROR
    23216 invalid bytes in program region, 5355 outside region.
    FAIL
    Memory did not validate. Did you erase?

Atmel DFU device disconnected (libusb0): Atmel Corp. ATmega32U4 (03EB:2FF4:0000)

Im not sure on what to do now as the lighting on the keyboard still works. Any help would be greatly appreciated on what I can do through QMK! Thank you!

kb-elmo, 2021-07-26:
> Not blank at 0x1
This indicates a corrupted bootloader on the controller of the board.
It's a known issue and unfortunately can only be fixed by ISP reflashing the bootloader.

KBDFans apparently made some mistakes when initially flashing the boards and setting the fuses of the controller.
This can result in the above error under certain circumstances.

For now there is no fix for it since it's not caused by QMK.

moyi4681, 2021-08-03:
it's using default bootloader, there is no reason to change the fuses of the conctrollder. I think it's a bug.

kb-elmo, 2021-08-03:
> it's using default bootloader, there is no reason to change the fuses of the conctrollder. I think it's a bug.

KBDfans PCBs are the only ones that have these problems.
If there was a bug in the Atmel-DFU bootloader other people would have these problems too but they don't.

So it has to be something specific to the hardware that KBDfans uses.

moyi4681, 2021-08-03:
https://kbdlab.co.kr/index.php?mid=board_CDST33&document_srl=5359619
https://geekhack.org/index.php?topic=88733.0
https://www.bountysource.com/issues/85387313-qmk-flashing-problem-memory-not-clearing
https://github.com/qmk/qmk_toolbox/issues/189
https://github.com/qmk/qmk_toolbox/issues/189
https://www.reddit.com/r/olkb/comments/d3820i/problem_with_flashing_not_blank_at_0x1/
https://homematic-forum.de/forum/viewtopic.php?t=56475

google not blank at ox01, there is so many other keyboard have the same problem. the schemic of atmega32u4 is simple, I don't think that is the hardware problem. if it something did cause it, we need to find out the reason

fauxpark, 2021-08-03:
Those are individual incidents, but the most consistent reports we've seen are from KBDFans boards. It could indicate a bad batch of chips.

closed 2022-04-17 by stale[bot]