qmk/qmk_firmware #13602: [Bug] Arch Linux qmk package gives AttributeError in script_qmk.py making it unusable

mcp292, 2021-07-19:
When I try to run `qmk flash -kb crkbd -km userme` I get this error:
```txt
Traceback (most recent call last):
  File "/usr/bin/qmk", line 33, in <module>
    sys.exit(load_entry_point('qmk==0.1.0', 'console_scripts', 'qmk')())
  File "/usr/bin/qmk", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.9/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/usr/lib/python3.9/site-packages/qmk_cli/script_qmk.py", line 18, in <module>
    milc.set_metadata(version=__version__)
AttributeError: module 'milc' has no attribute 'set_metadata'
```

I have successfully ran this command hundreds of times in the past.

I actually even get this with running `qmk`:
```txt
Traceback (most recent call last):
  File "/usr/bin/qmk", line 33, in <module>
    sys.exit(load_entry_point('qmk==0.1.0', 'console_scripts', 'qmk')())
  File "/usr/bin/qmk", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.9/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/usr/lib/python3.9/site-packages/qmk_cli/script_qmk.py", line 18, in <module>
    milc.set_metadata(version=__version__)
AttributeError: module 'milc' has no attribute 'set_metadata'
```

## System Information

 - Keyboard:  crkbd
 - Operating system: Arch Linux
 - AVR GCC version: 11.1.0
 - ARM GCC version: 11.1.0
 - QMK Firmware version: Should be [the latest](https://archlinux.org/packages/community/any/qmk/) but I get the above error when trying to run `qmk` or `qmk --version`.

martyschaer, 2021-07-25:
I was experiencing this as well.
It seems that I had milc 1.x.x installed using pip.
Removing that, and installing community/python-milc 1.4.2 was a workaround to this.

fauxpark, 2021-07-25:
Per https://docs.qmk.fm/#/newbs_getting_started?id=community-packages, the Arch `qmk` package is maintained by @Foxboron, not us.

Foxboron, 2021-07-25:
This isn't a packaging issue. Feel free to close and redirect them towards an Arch support channel.

>The current CLI version is 0.2.2, so this package is rather behind.

Yes, holiday and moving town the past 2 weeks.

https://lists.archlinux.org/pipermail/arch-dev-public/2021-July/030472.html

fauxpark, 2021-07-25:
You are right, this is not really a packaging issue since if the CLI is looking for `set_metadata` in MILC, the repo is even older. A simple `git pull` from the qmk_firmware directory will probably fix it up.

mcp292, 2021-08-02:
This was the actual solution:
https://github.com/qmk/qmk_firmware/issues/13602#issuecomment-886218586

Thanks @martyschaer!

closed 2021-08-02 by mcp292