qmk/qmk_firmware #13621: [Feature Suggestion] Adaptive keys

replicaJunction, 2021-07-20:
In some keyboard layouts, common bigrams (both in prose and in coding) are uncomfortable to type. For example, in QWERTY, it's rather uncomfortable to type the word "in" due to the row jump between the middle and index finger.

Adaptive keys are an idea that originated over at [/r/KeyboardLayouts](https://reddit.com/r/KeyboardLayouts). The idea is that by typing two keys in quick succession, the function of the second key can change based on the value of the first key. In this example (still in QWERTY), I might set up the letter U as an adaptive key, with the rule that "typing I-U sends I-N." Other fun examples could be Q-W to QU or H-J to HI.

Adaptive keys don't modify the behavior of the first key pressed - only the "adaptive" one changes its behavior.

I do have this feature working in [replicaJunction:feature_adaptive_keys](https://github.com/replicaJunction/qmk_firmware/tree/feature_adaptive_keys) (based on the `develop` branch), but I'd like to ask for feedback and review before I actually submit a PR.

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior
- [ ]

replicaJunction, 2021-07-24:
Updating to say that this is out of the "hey, here's a cool idea" stage and in the "here's some code for peer review" stage. The feature now even supports custom callbacks to inject your own code and logic if you want (modeled a bit after the Tap Dance code).

[Feature documentation](https://github.com/replicaJunction/qmk_firmware/blob/feature_adaptive_keys/docs/feature_adaptive_keys.md)

[Main logic](https://github.com/replicaJunction/qmk_firmware/blob/feature_adaptive_keys/quantum/process_keycode/process_adaptive_key.c)
* Still a lot of messy `#define` statements for now used for debugging. I anticipate removing the debug stuff soon.

The only change to the QMK core is to add a hook to `process_record_quantum`. It's entirely gated by a new makefile flag, so without setting `ADAPTIVE_KEY_ENABLE = yes` in a makefile, nothing new gets added to the build.

If anyone has time and interest and is willing to review, I'd appreciate it. I'm also around on Discord occasionally.

gkbd, 2021-07-25:
Looks promising, testing it in my layouts. Both simple ADAPTIVE_KEY and the ones with callbacks works fine so far.

replicaJunction, 2021-11-05:
There's been no activity on the issue because there's an open PR, #14034, pending review from QMK collaborators.

Seems to me like it would be appropriate to leave this issue open until the related PR is accepted or rejected. If it's rejected, there may be some additional discussion needed in this issue.

ariccb, 2022-02-16:
Is there a way we can get this un-staled? it's a feature i'd love to have!

replicaJunction, 2022-04-22:
@ariccb There didn't seem to be a lot of interest from the QMK core team, and since I developed it against the QMK dev branch, it kept running into merge conflicts that I had to go back and fix.

I've decided to implement this as a feature in my personal userspace instead of trying to merge it into QMK core.

You could also take a look at [this implementation over on /r/KeyboardLayouts](https://www.reddit.com/r/KeyboardLayouts/comments/smnv1o/adaptive_keys_qmk_implementation/). I actually think this might be a cleaner implementation of the same concept.

I'm happy to leave this issue open till I've got that implemented and merged, but it's really up to the QMK team whether to leave the issue open or close it at this point.

closed 2022-08-21 by github-actions[bot]