qmk/qmk_firmware #13661: [Bug] OSM and OSL not working on non-zero default layer

xsrvmy, 2021-07-23:
I am working on a one-handed layout, on which I am trying to use OSM for a sticky shift and OSL for a symbol layer. For obvious reasons this is not the default layer. I am running into an issue where neither OSM and OSL works when the one-handed layer is set to default (via DF(3)). OSM still works when held, but not when tapped. OSL does nothing at all.

xsrvmy, 2021-07-24:
Correction: Tested a bit more, and actually OSL(10) is sending a browser back if tapped (very bad), but holding works.

drashna, 2021-07-24:
OSM should definitely work in this case.  I use them a lot, and and can verify that. 

For OSL, that may be more complicated, and depends heavily on your keymap.