qmk/qmk_firmware #13662: Slow compilation speed on windows MSYS

xsrvmy, 2021-07-23:
I have recently switch back to using windows again and have downloaded the MSYS version of QMK. I have noticed that compilation is extremely slow. There is an extremely long delay between calling `qmk flash` and the message stating which layout is being compiled, and another shorter delay between that and the first file compilation message. Compilation itself also seemed to be slow. Is this expected because MSYS is slow, or is there a problem in the firmware?

daskygit, 2021-07-23:
This is pretty normal, if you want faster compilation try [QMK WSL](https://qmk.github.io/qmk_distro_wsl/). It's a little more complex to setup but works well.

zvecr, 2021-07-23:
> Is this expected because MSYS is slow, or is there a problem in the firmware?

Expected and partially being tracked in https://github.com/qmk/qmk_distro_msys/issues/23

xsrvmy, 2021-07-23:
Is WSL faster? And can you flash with WSL2 now or do I have to use QMK toolbox?

fauxpark, 2021-07-23:
WSL is much faster (it's essentially a Linux VM), as long as you keep the qmk_firmware repo inside the WSL filesystem: https://github.com/microsoft/WSL/issues/4197

Ciantic, 2021-07-24:
You can also use Docker to compile, and flash with teensy_loader_cli.exe (depending on your keyboard). 

I have example with that. It also includes a vscode .devcontainer environment, so if you use VSCode you can also use C++ environment with all the goodies. https://github.com/Ciantic/CianticQmkKeymap

xsrvmy, 2021-07-26:
Decided to setup QMK in WSL and yeah it is basically as fast as my linux boot. Only annoyance is having to flash with toolbox. I got VIA working though I can use that to fiddle around.

daskygit, 2021-07-26:
You should be able to flash most things using the QMK WSL distro I linked earlier.

drashna, 2021-07-31:
Also, make sure that you're using WSL version 2, if your system supports it.  v1 is okay, as it is faster than MSYS. However, version 2 is a LOT faster.

elfmimi, 2021-08-01:
Since while ago, savvy users have started using command like this. @xsrvmy 
`make preonic/rev3:default:flash DFU_UTIL=/mnt/c/QMK_MSYS/mingw64/bin/dfu-util.exe`
within arbitrary WSL2 environment + QMK_MSYS or QMK-Toolbox installed to the host environment.

drashna, 2021-08-09:
> Since while ago, savvy users have started using command like this. @xsrvmy
> `make preonic/rev3:default:flash DFU_UTIL=/mnt/c/QMK_MSYS/mingw64/bin/dfu-util.exe`
> within arbitrary WSL2 environment + QMK_MSYS or QMK-Toolbox installed to the host environment.
> 
> [#13431 (comment)](https://github.com/qmk/qmk_firmware/issues/13431#issuecomment-873855835)

That's ... effectively what QMK WSL is doing. 😆

closed 2022-04-17 by stale[bot]