qmk/qmk_firmware #13707: [Feature Request] Prettier make output with parallel build

Barabas5532, 2021-07-25:
Make output looks quite bad when using parallel build. For example, `qmk compile -j12 -kb lily58 -km barabas`:
[image]

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

RuRo, 2021-09-01:
@Barabas5532, FYI, I've just noticed, that I forgot to add the "fixes #13707" comment in my pull request. The feature you were asking for should be currently available on the devel branch.

tzarc, 2021-09-01:
Not for me. Latest `develop`:
[image]

RuRo, 2021-09-01:
Hm. Really weird. I'll take a closer look tomorrow. For now, can you tell me your `make --version`?

tzarc, 2021-09-01:
```
GNU Make 4.2.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

Barabas5532, 2021-09-03:
Thanks for the fix @RuRo!

Seems to work here on Arch Linux with make 4.3 (package version 4.3-3). Feel free to close this when @tzarc's issues are resolved.

RuRo, 2021-09-03:
@tzarc TL;DR: Try downgrading or upgrading your make version.

---

Okay, so this is really bizarre, but it seems to actually be a bug in make version `4.2.1`.
Consider the following simple `Makefile`:
```makefile
target: a b
	@echo "Done"

a:
	@echo a1
	@sleep 1
	@echo a2
	@sleep 1

b:
	@echo b1
	@sleep 1
	@echo b2
	@sleep 1
```
Running `make --jobs=16 --output-sync=target` is supposed to print
```
a1
a2
b1
b2
Done
```

But on Debian with `make 4.2.1` it prints
```
a1
b1
a2
b2
Done
```
instead. (notice that the `a` and `b` lines are interleaved)

I've tried different versions of make on different distributions and so far only `Debian Buster (10) + make 4.2.1-1.2` has this problem. The following distributions + make versions work as expected:
- `Debian Stretch (9) + make 4.1-9.1`
- `Debian Bullseye (11) + make 4.3-4.1`
- `Manjaro(Arch) + make 4.3-3`
- `Ubuntu 18.04 + make 4.1`

Unless I'm missing something, this is just a bug in this particular version of make and there's nothing we can do about it.

PR #13800 (2021-08-17): `--parallel` improvements
## Description

This PR has 2 improvements for `--parallel` support with make

1) `parallel` values less than 1 will now mean "unlimited" (see `make --help | grep -- --jobs`). This is not a breaking change, because currently non-positive `parallel` values fail with `make: the '-j' option requires a positive integer argument`.

2) If more than 1 job is used, the `--output-sync` option is also added. The stdout/stderr of each job will be collected by make and displayed synchronously, after each target is finished processing. This will fix #13707.

## Types of Changes

- [... cut]