qmk/qmk_firmware #13711: [Bug] MOUSEKEY_BASE_SPEED has no effect in Kinetic mode

y-usuzumi, 2021-07-25:
## Describe the Bug

In Kinetic mode, as per the document, `MOUSEKEY_BASE_SPEED` controls the maximum mouse speed. However, no matter which value I give it, the maximum mouse speed seems to be the same. However, if I increase MOUSEKEY_MOVE_DELTA, the maximum mouse speed increases.

My config:

```
#define MK_KINETIC_SPEED
#define MOUSEKEY_INTERVAL 8
#define MOUSEKEY_MOVE_DELTA 4
#define MOUSEKEY_INITIAL_SPEED 1
#define MOUSEKEY_BASE_SPEED 100
```

My goal is to increase the duration from pressing a mouse cursor button to reach its maximum speed.

Thanks!

## System Information

 - Keyboard:  Moonlander Mark I 
 - Operating system: Arch Linux
 - AVR GCC version: 11.1.0
 - ARM GCC version: 11.1.0
 - QMK Firmware version: 0.13.26
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

Lilalas, 2021-08-19:
@drashna Do you have any idea what’s going on here? I think I am affected by the same bug on my Planck EZ. I set my base speed to 0 for testing purposes, and my maximum speed is still extremely fast.

---

@y-usuzumi If all you want to archive is to increase the duration for reaching maximum speed, maybe setting `MOUSEKEY_MOVE_DELTA` to a lower value could help you. I don’t see why you would need to change `MOUSEKEY_BASE_SPEED` to archive this.

---

As a workaround for limiting the maximal speed, one can simply decrease the value of `MOUSEKEY_MOVE_MAX` instead of `MOUSEKEY_BASE_SPEED`. For example, add the following to your `config.h`:

```c
#define MOUSEKEY_MOVE_MAX 20
```

kuanyinglu, 2021-11-25:
I think the kinetic mode is still not working on v0.14.30.  In mousekey.c, the ifndef MK_COMBINED clause takes precedence over ifdef MK_KINETIC_SPEED. However, MOUSEKEY_MOVE_DELTA is still used by the other mode, which is why.

PR #17176 (2022-06-19): [Core] Mouse key kinetic mode fix

## Description

Iterating the mouse key kinetic mode fix from #16951 with functional mouse wheel and slightly faster and usable defaults.

## Types of Changes

- [x] Core
- [x] Bugfix
- [x] Documentation

## Issues Fixed or Closed by This PR

* fixes #13711

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_python)
- [x] I have read the [**PR Checklist** document](https://docs.qmk.fm/#/pr_checklist) and have made the appropriate changes [... cut]
closed 2024-10-15 by drashna