qmk/qmk_firmware #13743: [Bug] Cannot produce KC_NUBS keycode under Linux

mattdibi, 2021-07-27:
## Describe the Bug

I'm on a Corne keyboard and I cannot produce the `<`/`>` characters (italian layout) using `KC_NUBS` under Linux. It's always interpreted as `\`/`|`.

Likewise on macOS I can only produce `<`/`>` characters and no `\`/`|` (i.e. I can only send `KC_NUBS` but no `KC_GRV`)

## System Information

 - Keyboard:  Corne Chocolate
   - Revision (if applicable):  v2.1.0
 - Operating system: Linux/macOS
 - AVR GCC version: avr-gcc (Homebrew AVR GCC 8.4.0_2) 8.4.0
 - ARM GCC version: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
 - QMK Firmware version: 0.6.397 (also tried on master)
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

Please see attached modifications to the keymap I'm using (Miryoku layout)
[manna-harbour_miryoku copy.txt](https://github.com/qmk/qmk_firmware/files/6887450/manna-harbour_miryoku.copy.txt)

precondition, 2021-07-28:
What keyboard model is set in your OS? Make sure "Generic 105-key PC (intl.)" is set as your keyboard model in the keyboard settings.

If you use a DE that doesn't have a GUI app to configure keyboard settings, you can see the current keyboard settings with `setxkbmap -print` and change the keyboard model/geometry by running `setxkbmap -geometry 'pc(pc105)'`.

mattdibi, 2021-07-28:
Hi,

thank you for reaching out, unfortunately this didn't help much.

On macOS I was able to set it to ISO but now I cannot produce the `<`/`>` characters anymore as it's always interpreted as `\`/`|`.

On Linux it was already set as "Generic 105-key PC (intl.)"

mattdibi, 2021-07-28:
*Update*: I just flashed the [default crkbd keymap](https://github.com/qmk/qmk_firmware/tree/master/keyboards/crkbd/keymaps/default) on my Corne with `KC_NUBS` and `KC_GRV` where I could easily test them and they work without issue (i.e. I have both `<` and `\`).

Given the above I have to assume the problem comes from the fact that I'm using the Miryoku layout.

mattdibi, 2021-08-01:
Yeah so it always worked and I'm just dumb.

I mixed layers up and just put the same key in two different places -..-'

closed 2021-08-01 by mattdibi