qmk/qmk_firmware #13762: [Bug] Mouse keys cleared when changing layers

bmijanovich, 2021-07-28:
## Describe the Bug

Mouse key state is cleared when changing layers, which makes it impossible to hold KC_MS_BTN1 either physically or by use of `register_code16()` and `unregister_code16()` across layer changes.

## System Information

 - Keyboard:  Ploopy Classic
   - Revision:  1.005
 - Operating system: Mac OS Catalina 10.15.7
 - AVR GCC version: 
N/A, using QMK Docker container
 - ARM GCC version: 
N/A, using QMK Docker container
 - QMK Firmware version:
N/A, using QMK Docker contianer
 - Any keyboard related software installed? 
No

## Additional Context

I discovered this when trying to implement a "click and drag lock" feature, with the macro for the feature on a different layer than KC_MS_BTN1. When I released the layer, QMK would release KC_MS_BTN1 no matter what I did. This feature could be useful for either:

* People whose scroll wheel is operated by the same finger/thumb as KC_MS_BTN1
* People with a disability preventing holding a button down comfortably

drashna, 2021-08-09:
Can confirm the behavior, at least.

bmijanovich, 2023-12-19:
This was fixed a while back, mouse keys are no longer cleared on layer change.

closed 2023-12-19 by bmijanovich