qmk/qmk_firmware #13766: Added issues with new MCU models

joylee210, 2021-07-29:
I added a non-STM32 Cortex-M3 core MCU, but the keboards folder cannot contain McUconf. h file.
[image]

tzarc, 2021-07-29:
Please provide more information. You can add mcuconf.h.

joylee210, 2021-07-29:
The Mcuconf. h file ， included in keboards cannot be compiled at compile time，The McUconf. h file in the flatForms folder can be compiled

joylee210, 2021-07-29:
[image]

joylee210, 2021-07-29:
I don't know if I need to change the makefile when I add the MCU

joylee210, 2021-07-29:
[image]

joylee210, 2021-07-29:
With the exception of the McUconf. h file, QMK compacts properly and is usable

joylee210, 2021-07-29:
With the exception of the McUconf. h file, QMK can be compiled and the keyboard can work properly

joylee210, 2021-07-29:
[image]

joylee210, 2021-07-29:
Is it need to add MCU support anywhere else in your Python script file

fauxpark, 2021-07-29:
To add support for a new MCU (which is supported by ChibiOS), you will at least need to modify the following files/folders:

* `data/schemas/keyboard.jsonschema`
* `docs/compatible_microcontrollers.md` (add link to product page)
* `lib/python/qmk/constants.py`
* `platforms/chibios/` (add board files and default configs)
* `quantum/mcu_selection.mk`

There may be more modifications necessary in other files to support certain hardware features. Do note that if this MCU is *not* currently supported by ChibiOS we are unlikely to merge support for it here unless you first submit your contributions to ChibiOS itself.

joylee210, 2021-07-29:
@fauxpark  I've added all of that，Drivers have also been added to chiBIOS，The Mcuconf. h file in The Keyboards folder still needs to be included, but I haven't found the reason. I see that the path is included in the build Keyboard. Mk file, but it's still not compiled into it 
[image]

joylee210, 2021-07-29:
[image]

joylee210, 2021-07-29:
I don't understand why you need to include the Mcuconf.h file as well

fauxpark, 2021-07-29:
> I don't understand why you need to include the Mcuconf.h file as well

You don't. What is the error you're getting? You've just said "it cannot be compiled". That's not much to go on.

joylee210, 2021-07-29:
no error , I want to change the MCU configuration, but only mcuconf. h in Platforms folder is valid, and mcuconf. h in Mandolin. folder is invalid

joylee210, 2021-07-29:
Do you understand chinese? My English is so bad.
I want to be able to communicate in Chinese.

joylee210, 2021-07-29:
@fauxpark

fauxpark, 2021-07-29:
From what I can tell, your mcuconf.h is using the simplified format generated by `qmk chibios-confmigrate`. This contains the line `#include_next <mcuconf.h>`, which means there must exist a "full" mcuconf.h in `platforms/chibios` under the directory for this WB32 board, for example:
[image]

Your WB32F103 block in `mcu_selection.mk` must reference the name of this folder - in this case for `STM32F072`, there is the line `BOARD ?= GENERIC_STM32_F072XB`.

fauxpark, 2021-07-29:
Another thing I have noticed is that I don't see WB32 anywhere in:

https://github.com/ChibiOS/ChibiOS-Contrib/tree/chibios-20.3.x/os/hal/ports
https://github.com/ChibiOS/ChibiOS/tree/master/os/hal/ports

So I am assuming you have manually modified the `chibios` and `chibios-contrib` submodules inside `qmk_firmware/lib` to add support.

joylee210, 2021-07-29:
@fauxpark  My chiBIOS code has not been submitted, boards have been added in platforms/ CHIBIOS

joylee210, 2021-07-29:
[image]
[image]

joylee210, 2021-07-29:
[image]

fauxpark, 2021-07-29:
Okay, so you seem to know what you're doing then. I'm still not sure what you mean by this though:
> The Mcuconf. h file ， included in keboards cannot be compiled at compile time，The McUconf. h file in the flatForms folder can be compiled

Is the problem that the keyboard-level `mcuconf.h` is not overriding the platform-level one? ie. the `WB32_I2C_USE_I2C1 TRUE` line only works if you set it in `platforms/chibios/WB32_EVAL/configs/mcuconf.h`?

joylee210, 2021-07-29:
ye ， I don't know how to solve it

joylee210, 2021-07-29:
@fauxpark

joylee210, 2021-07-30:
Do you know what caused it @fauxpark

tzarc, 2021-08-02:
Delete the keyboard's `mcuconf.h` file.

joylee210, 2021-08-03:
@tzarc  Is there no other solution ?

tzarc, 2021-08-03:
Can you explain what you're trying to do, and why? There's no reason for the `mcuconf.h` in the keyboard if you've already got one in `platforms/chibios/XXXXXXXXXX/configs`.

joylee210, 2021-08-03:
@tzarc  I just want to be able to modify the configuration directly on the Keyboard's, so it's a little more convenient, and I've seen other Keyboards do that  .

tzarc, 2021-08-03:
I don't understand why you made all the other modifications, then. The makefiles and platforms are already set up to allow for that file to be included - no core makefile modifications are required, and the paths should be automatic.

I can't tell what the problem is because you've never pasted the error.

I'd suggest reverting any modifications done to any of the core makefiles -- unless absolutely necessary they won't get merged.

joylee210, 2021-08-03:
@tzarc  For the makefile, I only added the modifications for the MCU, nothing else was changed, and I don't know why mcuconf. h is not included. I also see that the makefile already includes the path.
The problem now is that I want to find out why mcuconf.h was not compiled when the makefile already included the path.

tzarc, 2021-08-03:
Can you do a build, and attach: `.build/obj_wb_keyboards_gsky_1_default/cflags.txt` please?

joylee210, 2021-08-03:
[cflags.txt](https://github.com/qmk/qmk_firmware/files/6920517/cflags.txt)
@tzarc

tzarc, 2021-08-03:
Given the parameters, it's already set up to include `mcuconf.h`.
You're going to have to paste the error -- there's nothing wrong with the build arguments.

joylee210, 2021-08-03:
[image]

@tzarc

joylee210, 2021-08-03:
path: keyboards\wb_keyboards\gsky_1\rev1\mcuconf.h

tzarc, 2021-08-03:
And what is being checked on:
`lib/chibios-contrib/os/hal/ports/WB32/LLD/I2Cv1/hal_i2c_lld.h`, line 236?

joylee210, 2021-08-03:
[image]
This is already enabled in mcuconf.h
@tzarc

joylee210, 2021-08-03:
This error occurs because the keyboards\wb_keyboards\gsky_1\rev1\mcuconf.h file has not been compiled into it
@tzarc

tzarc, 2021-08-03:
Please upload:
`keyboards/wb_keyboards/gsky_1/rev1/mcuconf.h` 
`platforms/chibios/WB32F103_EVAL/configs/mcuconf.h`
`lib/chibios-contrib/os/hal/ports/WB32/LLD/I2Cv1/hal_i2c_lld.h`

joylee210, 2021-08-03:
[hal_i2c_lld.txt](https://github.com/qmk/qmk_firmware/files/6920616/hal_i2c_lld.txt)
[platform_mcuconf.h.txt](https://github.com/qmk/qmk_firmware/files/6920617/platform_mcuconf.h.txt)
[keyboard_mcuconf.h.txt](https://github.com/qmk/qmk_firmware/files/6920618/keyboard_mcuconf.h.txt)
@tzarc

tzarc, 2021-08-03:
I can't tell from these files without being able to build myself.
I'd recommend pushing your branches of both qmk_firmware and ChibiOS-Contrib to personal forks, so we can investigate.

joylee210, 2021-08-03:
ok

joylee210, 2021-08-03:
Link to my warehouse : [https://github.com/joylee210/qmk_firmware.git  branch JoyLee](url) , branch JoyLee.
@tzarc

joylee210, 2021-08-03:
Link to my warehouse :   [https://github.com/joylee210/qmk_firmware.git](url)  , branch JoyLee , branch JoyLee.
@tzarc

joylee210, 2021-08-03:
Link to my warehouse : https://github.com/joylee210/qmk_firmware.git , branch JoyLee , branch JoyLee.
@tzarc

joylee210, 2021-08-04:
Did you figure out the structure ？ @tzarc

closed 2021-09-13 by joylee210