qmk/qmk_firmware #13782: [Bug] Massdrop CTRL Endgame - Build Error 'undefined reference to g_suspend_state'

SmokeMonsterPacks, 2021-07-30:
## Describe the Bug
Massdrop CTRL Endgame: Firmware building fails with the error 'undefined reference to g_suspend_state'

It can be resolved by deleting the reference to g_suspend_state from keymap.c and keymap.h in the Endgame files

It looks like it might be related to this (May 21 breaking changes): [https://github.com/qmk/qmk_firmware/commit/1646c0f26cfa21a7023d404008e4d0aa4917193d ](https://github.com/qmk/qmk_firmware/commit/1646c0f26cfa21a7023d404008e4d0aa4917193d )
One of the changes mentions USB Suspend changes, which points to this pull request: [https://github.com/qmk/qmk_firmware/pull/12697/files](https://github.com/qmk/qmk_firmware/pull/12697/files)

## System Information
 - Keyboard:  
   - Massdrop CTRL

## Additional Context
I had two friends confirm this, both on different flavors of Linux. 
[image]
[image]

tzarc, 2021-07-30:
Our `develop`-targeting CI doesn't build any keymaps other than `default` or `via`, so this wasn't going to get picked up unless someone explicitly built it.

Feel free to make a PR if what you've done fixes it.

SmokeMonsterPacks, 2021-07-30:
I'm afraid to mess something up with a PR, but here are the two fixed files if someone could please pull them in? 

I've simply removed the two references to g_suspend_state.

[image]
[Fixed Keymaps - Massdrop CTRL Endgame.zip](https://github.com/qmk/qmk_firmware/files/6910141/Fixed.Keymaps.-.Massdrop.CTRL.Endgame.zip)

PR #13805 (2021-07-31): Update endgame keymap to work with suspend changes
## Description

Updates the Massdrop CTRL keyboard endgame keymap to work after the changes to USB suspend in #12697

## Types of Changes

- [ ] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [x] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #13782

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/coding_conventions_ [... cut]
commit 3f5482bb (2021-08-02): Update endgame keymap to work with suspend changes (#13805)

Updates the Massdrop CTRL keyboard engame keymap to work after the
changes to USB suspend in #12697

Fixes #13782
closed 2021-07-31 by zvecr