qmk/qmk_firmware #13790: [Bug] Layer change not persistent and no startup sound when compiling under windows using QMK MSYS (planck/rev6)

jweickm, 2021-07-30:
Hi fellow planckers,

I have just gotten into customizing my planck/rev6 and it has been mostly a lot of fun, although now I have hit a wall. I have already spent many frustrating hours trying to get the startup sound back and to make the layer changes persistent. With the compiled keymap that I can download from within the QMK toolbox (from qmk.fm), everything works fine. However, when compiling on my machine, even when just compiling the "default" keymap that is found under "keymaps/default" I get no startup sounds and the layer changes are not persistent through power loss. 

To me, it seems that something must be wrong with the compiling environment, although there is no error message. Is the .bin file that I can download from qmk.fm directly generated based on the files in "keymaps/default" or am I missing something?

Does somebody else experience the same issue when compiling under windows 10? I have followed the instructions to the letter and used the one-click QMK  MSYS installer from `https://msys.qmk.fm/`. 

Btw, compiling from the online configurator also resulted in me losing the startup sounds and the layer persistence. 

I am grateful for any advice on how to proceed from here. 
Thanks a lot, 
J

## Describe the Bug

- When compiling the default keymap under windows 10 with QMK MSYS:
- Layer changes not persistent
- No startup sounds

## System Information

 - Keyboard:  PLANCK
   - Revision (if applicable):  REV6
 - Operating system: WINDOWS 10
 - AVR GCC version:  8.4.0

 - ARM GCC version:  10.1.0

 - QMK Firmware version: 0.13.19

 - Any keyboard related software installed? 
   - [x] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

drashna, 2021-08-09:
This is a known issue, at least the audio part. 
https://github.com/qmk/qmk_firmware/pull/13606

The persistent layers, are you using the `set_single_persistent_default_layer` function?  If not, then they won't be persistent.

jweickm, 2021-08-09:
> This is a known issue, at least the audio part.
> #13606
> 
> The persistent layers, are you using the `set_single_persistent_default_layer` function? If not, then they won't be persistent.

Thank you for linking to the audio bug. I was not aware of that other issue. 

Regarding the persistent layer change, I actually am using `set_single_persistent_default_layer` just as in the default keymap.c. 
I attached my keymap.c (changed the file ending to .txt as Github does not "support" .c files) for further reference. I usually do not expect someone else to implement some feature for me, but I am really at a loss here, and would appreciate some pointers. From what I have read, my keymap.c should make layer changes persistent. 
Thanks already in advance to anybody taking a look at it. 
j

[keymap.txt](https://github.com/qmk/qmk_firmware/files/6953397/keymap.txt)

jweickm, 2021-08-09:
Interestingly, I got it to work now!! Unfortunately, I cannot exactly say what solved it in the end, but I did disable 
```
CONSOLE_ENABLE = no         # Console for debug
BOOTMAGIC_ENABLE = no      # Virtual DIP switch configuration
```
in the rules.mk file, which were enabled by default. It seems that the enabled bootmagic prevented the function `set_single_persistent_default_layer` to do its magic. 
I would close this issue now, and hope that maybe some others will find this helpful. 
I think that disabling `BOOTMAGIC_ENABLE` by default would also be a good option.

closed 2021-08-09 by jweickm