qmk/qmk_firmware #13816: QMK make finished with error "TIM2 is not a 32bits timer"

AkiTsuki114514, 2021-07-31:
I tried to build firmware for my board with stm32f103cbt6, but it failed with following errors
```
Compiling: lib/chibios/os/hal/ports/STM32/LLD/TIMv1/hal_st_lld.c                                   lib/chibios/os/hal/ports/STM32/LLD/TIMv1/hal_st_lld.c:48:2: error: #error "TIM2 is not a 32bits timer"
   48 | #error "TIM2 is not a 32bits timer"
      |  ^~~~~
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:436: .build/obj_shiro/lib/chibios/os/hal/ports/STM32/LLD/TIMv1/hal_st_lld.o] Error 1
Make finished with errors
```
in the rules.mk
```
MCU = STM32F103
BOOTLOADER = stm32duino
```
It can compile success with other MCU
I tried other koyboards, and they were failed with STM32F103

zvecr, 2021-07-31:
Need a `chconf.h` like <keyboards/handwired/onekey/bluepill/chconf.h>, setting `#define CH_CFG_ST_TIMEDELTA 0`

closed 2021-07-31 by AkiTsuki114514

thzx805, 2022-03-17:
> Need a `chconf.h` like https://github.com/qmk/qmk_firmware/tree/master/keyboards/handwired/onekey/bluepill/chconf.h, setting `#define CH_CFG_ST_TIMEDELTA 0`

May I know how to settting this ,I met the same error